compas-dev/compas #545: Box.from_width_height_depth orientation

joburger, 2020-05-15:
It seems like Box.from_width_height_depth does not position the bottom left corner of the box at the origin of the coordinates system, but the center point of the box just like the standard Box() method, even though it says so in the docs: https://compas-dev.github.io/main/api/generated/compas.geometry.Box.from_width_height_depth.html#compas.geometry.Box.from_width_height_depth 

**To Reproduce**
```
box = Box.from_width_height_depth(1.0, 2.0, 3.0)
```

**Expected behavior**
Expected the bottom left corner of the box to be in (0, 0, 0) but actually it is in (-0.5, -1.5, -1.0)

**Screenshots**
[image]

brgcode, 2020-05-15:
hi joris,

the box is in fact supposed to be centered at the origin. so the behaviour is correct, but the docs are wrong. will fix the docs now...

joburger, 2020-05-15:
ok thx!

brgcode, 2020-05-15:
should be fixed since c4f53a24

closed 2020-05-15 by brgcode