compas-dev/compas #548: Compas non-manifold but valid mesh crashes

ioanna21, 2020-05-23:
**Describe the bug**
I am creating some meshes by splitting a bigger mesh. The resulting meshes are valid, but non-manifold. When I try to operate on these meshes (for example to iterate through the boundary edges), the application freezes and slowly eats up all the memory, until it crashes a few minutes later with a 'Bad memory allocation'.  

**To Reproduce**
I created a sample script that reproduces the crash. When you run it, open the task manager to observe the memory usage. You can download it here.
https://www.dropbox.com/sh/uipx8gl234p9nmi/AAA2tKOcy25DR0gDJJ85Af5na?dl=0

**Expected behavior**
It should print out the edges that are on the boundary. Instead it crashes after using all the available memory. 

**Desktop (please complete the following information):**
 - OS:  Windows 10
 - Python version: Python 3.7.6
 - Python package manager: conda

Compas version :  0.15.6 (latest)

brgcode, 2020-05-27:
the algorithm gets confused because of that single vertex connection separating two of the boundary "openings". if i remove it everything works fine. not sure if this behaviour is wrong but i will see what i can do about it...

ioanna21, 2020-05-28:
The problem is that the behavior is not consistent. So I create many of those meshes and in most cases it works well, but with some it crashes. If the problem is in the geometry, I was wondering how I can detect it in advance so that I avoid the crash. Does compas in general have issues with non-manifold meshes or this is a particular case?

brgcode, 2020-05-28:
non-manifold meshes can be stored in the data structure, but they cannot be navigated because they don't have a proper halfedge representation, which is what the topology of the mesh is based on. i would have to check but i think there is `is_manifold` or `is_orientable` which should allow you to find out if the mesh is a proper halfedge. otherwise we can add it :)

ioanna21, 2020-05-29:
The `is_manifold()` works. The `is_orientable()` raises a NotImplemetedError. That is why I used the `unify_cycles()` first which works (most of the time) on the non-manifold meshes. I thought that if this work then the mesh should be orientable and traversable. So, just to make sure I understood, if I want to traverse meshes like this one, I would have to remove the vertices that make them non-manifold.

brgcode, 2020-11-23:
can we close this?

closed 2020-12-01 by brgcode