compas-dev/compas #552: mesh.vertex_faces() - not all faces lister

funkchaser, 2020-06-02:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Context [e.g. ST3, Rhino, Blender, ...]
2. Sample script
3. Sample data
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Python version [e.g. 2.7]
 - Python package manager [e.g. macports, pip, conda]

**Additional context**
Add any other context about the problem here.

brgcode, 2020-06-03:
the cycles of the polygons are not consistent.
unifying the cycles after constructing the mesh seems to solve the problem...

```python
from compas.datastructures import Mesh
from compas_plotters import MeshPlotter

polygons = [
    [[6.509477157600335, -6.7199101640508907, 0.0], [3.0969329040796705, -8.0661432181920709, 0.0], [3.3160871221956763, -11.165610017261301, 0.0], [5.5445756291656991, -12.655949856690249, 0.0], [8.043556684412378, -11.541302962603025, 0.0], [8.8575580659861153, -8.7236058725400891, 0.0]],
    [[6.509477157600335, -6.7199101640508907, 0.0], [8.9561875348052826, -4.3570303399387011, 0.0], [8.8575580659861153, -8.7236058725400891, 0.0]],
    [[8.8575580659861153, -8.7236058725400891, 0.0], [11.675255156049049, -9.6315304904492578, 0.0], [12.86494948296451, -7.0956031093926155, 0.0], [8.9561875348052826, -4.3570303399387011, 0.0]],
    [[8.043556684412378, -11.541302962603025, 0.0], [8.8575580659861153, -8.7236058725400891, 0.0], [11.675255156049049, -9.6315304904492578, 0.0], [12.991124816051649, -11.967308968323813, 0.0], [9.1706355204375516, -14.171153579995098, 0.0]]]

mesh = Mesh.from_polygons(polygons)
mesh.unify_cycles()

for key in mesh.vertices():
    print("vertex", key, "-> faces:", mesh.vertex_faces(key))

plotter = MeshPlotter(mesh, figsize=(8, 5))
plotter.draw_vertices(text='key', radius=0.25)
plotter.draw_faces(text='key')
plotter.show()
```

funkchaser, 2020-06-09:
yes it works now, thanks. 
Is it intended for the user to take care of it explicitly? If yes, then I'd suggest to note this in the API.

brgcode, 2020-06-09:
yes, it is. will make a not in the docs.

closed 2020-06-09 by brgcode