compas-dev/compas #554: export multiple meshes in a single obj

Achillx, 2020-06-04:
# Feature Request

It would be useful to be able to export a list of COMPAS meshes in a single obj file.

## Details

Currently, all exporters work on single objects with a `.to_obj()` method. 

I've created a derivative function from `.to_obj()` found inside the mesh class. It does what I've described above, but I don't know where is the right place to add such a multi-object exporter.

```python
def export_meshes_to_obj(meshes, filepath):
    """Write the mesh to an OBJ file.

    If the obj already contains mesh data, your data will be appended at the end.

    Parameters
    ----------
    meshes : list of :class:`compas.datastructures.Mesh`
        The meshes you want to export.
    filepath : str
        Full path of the file.

    """
    # clean the file
    with open(filepath, "w") as fh:
        fh.truncate(0)

    # obj vertices are based on their line number
    # inside the file and they don't start with 0
    last_vkey = 1

    for index, mesh in enumerate(meshes):
        # get the name of the mesh
        mesh_name = mesh.attributes.get('name')
        # When you import the obj to another software all meshes with the
        # same name will be seen as one. So we rename the default name.
        if mesh_name == "Mesh":
            mesh_name = "Mesh" + str(index)

        key_index = mesh.key_index()

        with open(filepath, 'a') as fh:
            fh.write('o {}\n'.format(mesh_name))
            for _key, attr in mesh.vertices(True):
                fh.write('v {0[x]:.3f} {0[y]:.3f} {0[z]:.3f}\n'.format(attr))
            for fkey in mesh.faces():
                vertices = mesh.face_vertices(fkey)
                vertices = [key_index[key] + last_vkey for key in vertices]
                fh.write(' '.join(['f'] + [str(index)
                         for index in vertices]) + '\n')

        last_vkey += mesh._max_int_key + 1

if __name__ == "__main__":
    import os
    from compas.geometry import Box
    from compas.datastructures import Mesh

    HERE = os.path.dirname(__file__)

    a = Box.from_width_height_depth(5.0, 3.0, 1.0)
    b = Box.from_width_height_depth(1.0, 5.0, 3.0)
    c = Box.from_width_height_depth(3.0, 2.0, 7.0)

    a = Mesh.from_shape(a)
    a.attributes['name'] = 'mesh_1'
    b = Mesh.from_shape(b)
    c = Mesh.from_shape(c)

    meshes = [a, b, c]

    filename = os.path.join(HERE, 'meshes.obj')
    export_meshes_to_obj(meshes, filename)
    print("Meshes exported at:", filename)

```

brgcode, 2020-06-04:
okay, i guess we could add a static method on the mesh class for that...

yck011522, 2020-07-12:
On the same note. I believe it would be nice for all default `compas` class to comply with some standard python JSONEncoder. Such that any collections of compas objects, or user classes that contains compas objects can automatically and recursively serialize them all.

 I do not know how that works in Python. But I believe it is about subclassing the correct class and implement/override certain methods for de/serialization. In .net environment it is very easy to implement the ISerializable interface.

It would be nice to be able to just `json.dump(any_collection_of_compas_objects)`

brgcode, 2020-07-13:
@yck011522 this is indeed relevant. however, please file as separate issue or request...

beverlylytle, 2021-07-30:
@brgcode This is solved, no?

brgcode, 2021-07-30:
yes it is

beverlylytle, 2021-07-30:
Then I guess I'll close it.

closed 2021-07-30 by beverlylytle