compas-dev/compas #555: Add unweld option when exporting meshes in obj

Achillx, 2020-06-04:
# Feature Request

When exporting a mesh to obj choose to have it unwelded or welded.

## Details

When quickly previewing welded meshes with mac's built-in preview, or when importing them in Rhino, they look horrendous!

Allow passing welded or unwelded to mesh exporter. `mesh.to_obj(welded=False)`

[image]
Top 3 meshes are just imported and the bottom ones have been unwelded using Rhino's `Unweld` command.

[image]
Here a closer look.

brgcode, 2020-06-04:
will include in the next release...

brgcode, 2020-06-04:
the latest dev version should have an unweld option.
please check...

brgcode, 2020-06-11:
@Achillx ping?

Achillx, 2020-06-11:
Sorry, I'm still working on an older version of compas.

- I made an environment with compas version 0.16.0-74777898
- Loaded a welded obj with `mesh.from_obj()`
- exported it again welded and unwelded

The unwelded version looks perfect!

closed 2020-06-11 by Achillx