compas-dev/compas #560: mesh.transformation() massively slower in 0.16.0 when used in Rhino Grasshopper

yck011522, 2020-06-08:
**Describe the bug**
After the upgrade from 0.15.6 to 0.16.1. 
Performing a `mesh.transformation()` is orders of magnitude slower from Rhino Grasshopper.

### Tests performed in cpython (no problem with both versions)
Performed with python=3.7.6 
compas=0.15.6 vs 0.16.1 (both versions have similar results)
```
import cProfile

import compas
from compas.datastructures import Mesh, mesh_transform
from compas.geometry import Frame, Point, Transformation, Vector

print('compas.__version__ : ', compas.__version__)

f1 = Frame([2, 2, 2], [0.12, 0.58, 0.81], [-0.80, 0.53, -0.26])
f2 = Frame([1, 1, 1], [0.68, 0.68, 0.27], [-0.67, 0.73, -0.15])
T = Transformation.from_frame_to_frame(f1, f2)
mesh = Mesh.from_ply(compas.get_bunny())

print('mesh.number_of_vertices() : ', mesh.number_of_vertices())

def transforamtion_test():
    for _ in range(1):
        mesh_transform(mesh, T)

cProfile.run('transforamtion_test()', sort=2)
```

[image]

Result = 0.55 vs 0.60 seconds. I'd say not much difference

### Test performed in Rhino GH + compas 0.15.6 (no problem)

[image]

Result = 0.65 seconds. Very similar to cPython speed

### Test performed in Rhino GH + compas 0.16.1 (very slow)

[image]

Result =140.7 seconds. **This is more than x100 times slower!!!**

### Perhaps of interest

What is more interesting is the following scenario. When I take away the `unload_modules()` line from the beginning of my GH script. (Which is the my typical use case) And then re-run the same operation. The run-time is even more dramatic (470 seconds!!!).

[image]

This smells like a memory leak to me somewhere in the IronPython implementation that is triggered by the new transformation functions in compas 0.16.

### Additional context
This issue is posted after 2 hours of debugging with @gonzalocasas : about why RobotModel being very slow when I try to draw the meshes.

This problem cause a big problem for my GH scripts where I have 10 to 20 meshes to move around. 

[tests.zip](https://github.com/compas-dev/compas/files/4748381/tests.zip)

brgcode, 2020-06-09:
just out of curiosity, could you run your test in Rhino (w/o GH) just to make sure the problem is related to ipy and not GH?

thanks

brgcode, 2020-06-09:
could you try removing this `or self.defaults[name]` from line 33 in `compas.datastructures.mesh.core.attributes.py`?

yck011522, 2020-06-09:
I tested the following script [test_rhinopy_mesh_transform_problem.zip](https://github.com/compas-dev/compas/files/4751716/test_rhinopy_mesh_transform_problem.zip) in Rhino Python Editor (no Grasshopper involved here). But the results are not promising. 

```
import compas
from compas.datastructures import Mesh, mesh_transform
from compas.geometry import Frame, Point, Transformation, Vector
print('compas.__version__ : ', compas.__version__)

f1 = Frame([2, 2, 2], [0.12, 0.58, 0.81], [-0.80, 0.53, -0.26])
f2 = Frame([1, 1, 1], [0.68, 0.68, 0.27], [-0.67, 0.73, -0.15])
T = Transformation.from_frame_to_frame(f1, f2)
mesh = Mesh.from_ply(compas.get_bunny())

print('mesh.number_of_vertices() : ', mesh.number_of_vertices())

def transforamtion_test():
    for _ in range(1):
        mesh_transform(mesh, T)

import time
s = time.time()
transforamtion_test()
print (time.time() - s)
```
### Test 1
Result for `compas.datastructures.mesh.core.attributes.py` un-changed:
```
('compas.__version__ : ', '0.16.1')
('mesh.number_of_vertices() : ', 35947)
83.1322937012
```
### Test 2
Result for `compas.datastructures.mesh.core.attributes.py` line 33 changed as you suggested:
```
('compas.__version__ : ', '0.16.1')
('mesh.number_of_vertices() : ', 35947)
104.823860168
```
Second run without "reset script engine" :
```
('compas.__version__ : ', '0.16.1')
('mesh.number_of_vertices() : ', 35947)
465.088577271
```
I think there is still a smell of some leaks or something get accumulated.

brgcode, 2020-06-09:
i have the feeling this will solve the problem #561
if you could try/test that would be great...

gonzalocasas, 2020-06-09:
Just tested ironpython without rhino (ipy directly on the command line), on current master and also #561 and it does not look like it's fixed yet, but I was not patient enough to wait 7 minutes for it to finish. I will try to debug a bit more in detail and report back.

brgcode, 2020-06-09:
i pushed an update. if that doesn't solve it, it is not related to the attribute views...

yck011522, 2020-06-09:
I just tested again with the same test file [test_rhinopy_mesh_transform_problem.zip](https://github.com/compas-dev/compas/files/4751716/test_rhinopy_mesh_transform_problem.zip) in Rhino after pulling commit e102be36d78fa8808459327faef6356997dedd80

Problem still not resolved. 
```
('compas.__version__ : ', '0.16.1')
('mesh.number_of_vertices() : ', 35947)
127.443771362
```

brgcode, 2020-06-09:
and you are sure that you are using the code of the commit? because it should show up in the version info...

gonzalocasas, 2020-06-09:
yep, still super slow.... I am running it and still hasn't finished:
```
(compas-fab-38) C:\Users\gcasas\eth\Projects\research\compas_fab\tests>ipy test_mesh_perf.py
('compas.__version__ : ', '0.16.0-e102be36')
('mesh.number_of_vertices() : ', 35947)
```

I also ran the same test but modified the `mest_transform` function to NOT assign the the results of the transformation back to the vertices (ie the `for` loop right after transforming points), and the result actually finished before I got annoyed, it's slow, but I am pretty sure the problem is in the step after the transformation.
```
(compas-fab-38) C:\Users\gcasas\eth\Projects\research\compas_fab\tests>ipy test_mesh_perf.py
('compas.__version__ : ', '0.16.0-e102be36')
('mesh.number_of_vertices() : ', 35947)
25.020
```

gonzalocasas, 2020-06-09:
Additionally, if I simply do this after the mesh transform:
```python
for index, (key, attr) in enumerate(mesh.vertices(True)):
    pass
```

which should be extremely lightweight, the total time shoots up.

brgcode, 2020-06-09:
can you try this
```python
for index, key in enumerate(mesh.vertices()):
    mesh.vertex_attributes(key, 'xyz', xyz[index])
```

gonzalocasas, 2020-06-09:
I just tried something similar. I tested only performance, and it works very fast, but didn't check correctness:
```python
def mesh_transform(mesh, transformation):
    vkeys = list(mesh.vertices())
    vertices = [mesh.vertex_coordinates(key) for key in vkeys]
    xyz = transform_points(vertices, transformation)
    for key in vkeys:
        mesh.vertex_attributes(key, 'xyz', xyz[key])
```

It looks similar to your stuff, but not quite, so if you could check correctness, it'd be cool.

I mean, that is so fast, that I'm curious if it's actually working:
```
(compas-fab-38) C:\Users\gcasas\eth\Projects\research\compas_fab\tests>ipy test_mesh_perf.py
('compas.__version__ : ', '0.16.0-e102be36')
mesh loading: 3.82s
('mesh.number_of_vertices() : ', 35947)
mesh transform: 1.53s
```

brgcode, 2020-06-09:
that looks correct to me

gonzalocasas, 2020-06-09:
yup, just checked visually inside gh and it looks good :)

gonzalocasas, 2020-06-09:
[image]

brgcode, 2020-06-09:
that is very nice, but we should still find the source of the problem, no?

gonzalocasas, 2020-06-09:
yes, agree.
However, we could apply this fix in the `mesh_transform` anyway, as it is a more efficient way to do it.

brgcode, 2020-06-09:
okay. will do. i will immediately also change the attribute thing because it is in any case not correct now...

brgcode, 2020-06-09:
this also seems like a perfect example of not really requiring a data structure, but only a triangle soup. that would speed up the loading time quite a bit and with transformation methods like `transform_cli` and `transform_numpy` also the transformation part...

gonzalocasas, 2020-06-09:
I am not sold on the "non-datastructure" idea, but could be explore specific implementation options...

For instance, C# has some classes that have a regular version and a `Slim` variant, eg. `ManualResetEventSlim` and `ManualResetEvent`. Both have relatively similar APIs, but their implementations differ enough internally that they are different classes.  Perhaps we could try a similar approach. That would, however, further blur the distinction of what is a data structure and what is not.

brgcode, 2020-06-09:
not sure we are talking about the same thing. maybe it is better to base the discussion on an example. i have made a `TriMesh` in/for `compas_cgal` so we could use that (will push soon). the performance is orders of magnitude different...

brgcode, 2020-06-09:
should be solved in 6f6704c7

yck011522, 2020-06-09:
I can confirm that the transform function is fixed. 
I tested with the same Rhino Python Script and the results are comparable with CPython speed.
```
0.16.0
('mesh.number_of_vertices() : ', 35947)
0.751541137695
```

I also tried the transformation functions that were called from a RobotArtist, and the behavior returned back to normal. I think this issue can be closed if you agree @brgcode 

Thank you very much for the fast response in fixing this issue.

brgcode, 2020-06-10:
happy to hear its working and we can close indeed, but we should make a note somewhere or file a new issue, or organise a bug hunt to figure out what is wrong with the attribute view. what do you think @gonzalocasas ?

gonzalocasas, 2020-06-10:
yep, closing this. I'll create a follow up for the vertex attribute view.

closed 2020-06-10 by gonzalocasas