compas-dev/compas #562: Diagnose performance issues with vertex attribute view

gonzalocasas, 2020-06-10:
During the resolution of #560, we noticed the vertex attribute view object has some severe performance issues when running on IronPython. The issue was resolved side-stepping the view, but the underlying problem of this class should be diagnosed and resolved.

**To Reproduce**

1. Open Rhino editor or run the following script using IronPython CLI (`ipy`):
```python
from timeit import default_timer as timer
import compas
from compas.datastructures import Mesh

t0 = timer()
mesh = Mesh.from_ply(compas.get_bunny())
print('mesh load time: {:.2f}s'.format(timer() - t0))

t1 = timer()
for _ in mesh.vertices(False):
    pass
print('vertex iterator without attribute view: {:.2f}s'.format(timer() - t1))

t2 = timer()
for _ in mesh.vertices(True):
    pass
print('vertex attribute view iterator: {:.2f}s'.format(timer() - t2))
```
which will output something like:
```
mesh load time: 2.26s
vertex iterator without attribute view: 0.03s
vertex attribute view iterator: 125.96s
```

For reference, the exact same script outputs the following when executed using cpython:
```
mesh load time: 2.09s
vertex iterator without attribute view: 0.01s
vertex attribute view iterator: 0.13s
```

**Expected behavior**
The run time of iterating vertices with attributes should **not be 4000x slower** 😛 

**Desktop (please complete the following information):**
 - OS: Windows
 - Python version: IronPython 2.7.8 / Rhino 6
 - Python package manager: `conda`
 - compas version: `0.16.1`

gonzalocasas, 2020-06-10:
The problem seems to come from `MutableMapping`.
Apparently, the IronPython implementation sucks real bad.
@brgcode what do we use `MutableMapping` for?

brgcode, 2020-06-10:
proper form i guess :)

brgcode, 2020-06-10:
it also provides a bit of base functionality that we don't really use but that one might expect to get from a dict

commit 13a71a14 (2020-06-11): fix #562: add vendered copies of MutableMapping and Mapping without ABCMeta as metaclass
commit 74777898 (2020-06-11): Merge pull request #563 from compas-dev/fix-mutable-mapping

fix #562: fix attribute view iteration performance issues
closed 2020-06-11 by brgcode