compas-dev/compas #564: RobotModel.to_data()

yck011522, 2020-06-17:
# Feature Request

As a end-user, I want to serialize a RobotModel that I created programatically so that I can load it back later from file.

## Details

I have a lot of trouble trying to pickle `RobotModel`, either using `pickle` or `jsonPickle` with no success. The decoded object doesn't function anymore.

I created a little script to check if my returned object still works:

```
def LoadToolAndCheck(json_path):
    with open(json_path, 'r') as f:
        json_str = f.read()
        tool = jsonpickle.decode(json_str, keys=True)

    print (tool.__class__)
    for l in tool.links:
        print (l.__class__, l)
```
One specific scenario is when I create this `RobotModel` object in Rhino GH, and then I deserialize it in CPython, it will not function anymore.

**Describe the solution you'd like**
I want a way to serialize RobotModel objects, either by `jsonpickle` or `pickle` or a 'to_data' type of method. I believe for the consistency with other compas object, a `to_data()` method is most appropriate.

brgcode, 2020-06-18:
`RobotModel` does not have a `to_data` method?

yck011522, 2020-06-19:
@brgcode  In case that is not a rhetorical exclamation 😅  . Then the answer is:

`RobotModel` does not have a `to_data` method.

yck011522, 2020-06-23:
Hello. @gonzalocasas  @brgcode  Just wonder if there are any new thoughts about the issue and if there are some quick workarounds. 

Workaround potential: I need to serialize only a model that has 1 planning group. I have a few links and a few joints, created programmatically. All have attached compas mesh as their visual mesh.

brgcode, 2020-06-24:
sorry forgot about this. in my opinion, `RobotModel` should definitely have `from/to_data` methods... if @gonzalocasas has the time i would prefer deferring this to him :)

gonzalocasas, 2020-06-24:
yes, it should have them. I will work on this soon in the short term.

closed 2020-07-24 by beverlylytle