compas-dev/compas #566: Compas for Rhino7

ioanna21, 2020-06-24:
# Feature Request

I was wondering if there is a way to install compas on Rhino7. I haven't managed to find something relevant. Let me know if there is some solution for that available, or if it is being planned for some time in the future

Thanks,
Ioanna

brgcode, 2020-06-24:
there should not be an issue with this. have you tried? did it not work?

brgcode, 2020-06-24:
you do have to specify the version when installing (6.0 is the default).

```bash
python -m compas_rhino.install -v 7.0
```

ioanna21, 2020-06-24:
Thanks for the quick reply! This gives me the error
```

usage: install.py [-h] [-v {5.0,6.0}] [-p PACKAGES [PACKAGES ...]]
install.py: error: argument -v/--version: invalid choice: '7.0' (choose from '5.0', '6.0')
```

brgcode, 2020-06-24:
sorry forgot to add the option to the command line interface. could you update compas and try again?

ioanna21, 2020-06-24:
Great, it works! Thanks a lot!

closed 2020-06-24 by ioanna21