compas-dev/compas #567: Problem while loading with Mesh.from_json()

salmamzfr, 2020-06-24:
**Describe the bug**
Mesh.from_json() does not work if the the file is saved using a different COMPAS version.

**To Reproduce**
The following error shows up in Ipython:

```
Message: malformed string

Traceback:
line 79, in _convert, “C:\Program Files\Rhino 6\Plug-ins\IronPython\Lib\ast.py”
line 80, in literal_eval, “C:\Program Files\Rhino 6\Plug-ins\IronPython\Lib\ast.py”
line 302, in data, “C:\Users\msalma\AppData\Roaming\McNeel\Rhinoceros\6.0\Plug-ins\IronPython (814d908a-e25c-493d-97e9-ee3861957f49)\settings\lib\compas\datastructures\mesh\core\halfedge.py”
line 376, in from_json, “C:\Users\msalma\AppData\Roaming\McNeel\Rhinoceros\6.0\Plug-ins\IronPython (814d908a-e25c-493d-97e9-ee3861957f49)\settings\lib\compas\datastructures\mesh\core\halfedge.py”
```

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Python version [e.g. 2.7]
 - Python package manager [e.g. macports, pip, conda]

**Additional context**
Add any other context about the problem here.

gonzalocasas, 2020-06-24:
@salmamzfr do you remember which version was it saved with? and which version are you reading it with now?

brgcode, 2020-06-24:
@gonzalocasas recent versions of the mesh data structure use exclusively int keys. therefore it is no longer necessary to use `repr` and `literal_eval`. the current version of the mesh uses `str` and `int` instead...

nizartaha, 2020-06-24:
@gonzalocasas i get the same error saving the mesh from compas `0.16.1` and reading in compas `0.15.6`

gonzalocasas, 2020-06-24:
> recent versions of the mesh data structure use exclusively int keys. therefore it is no longer necessary to use `repr` and `literal_eval`. the current version of the mesh uses `str` and `int` instead...

which reminds me we should prioritize #520

salmamzfr, 2020-06-24:
> @salmamzfr do you remember which version was it saved with? and which version are you reading it with now?

I think it was saving with 0.16.0 and reading 0.15.6 or the other way around!

salmamzfr, 2020-06-24:
> > @gonzalocasas  do you remember which version was it saved with? and which version are you reading it with now?
> 
> I think it was saving with 0.16.0 and reading 0.15.6 or the other way around!

brgcode, 2020-07-06:
could you make the JSON available somewhere? i will try to add a fix for this, but i would like to make sure i am fixing the correct problem...

nizartaha, 2020-07-06:
here is the mesh saved from compas `0.16.1`
[mesh_NT.zip](https://github.com/compas-dev/compas/files/4878663/mesh_NT.zip)
Thanks

brgcode, 2020-10-28:
@salmamzfr in recent versions this compatibility problem should be solved. can you check if this issue can be closed?

salmamzfr, 2020-11-10:
Hi Tom, 

I am working with the latest compas version from compas source: '0.17.2-c422420c', when I save a mesh (mesh.to_json) in python, I cannot load the mesh in Ironpython, it gives me the following error, which is the same as before:

Message: attributes

Traceback:
  line 174, in data, "C:\Users\msalma\AppData\Roaming\McNeel\Rhinoceros\6.0\Plug-ins\IronPython (814d908a-e25c-493d-97e9-ee3861957f49)\settings\lib\compas\datastructures\mesh\core\halfedge.py"
  line 275, in from_json, "C:\Users\msalma\AppData\Roaming\McNeel\Rhinoceros\6.0\Plug-ins\IronPython (814d908a-e25c-493d-97e9-ee3861957f49)\settings\lib\compas\datastructures\mesh\core\halfedge.py"

salmamzfr, 2020-11-10:
Also, I have been trying a similar process with network.to_json in python and loading it in Ironpython (network.from_json), I am able to load it, but it gives me the an empty network. Here is the print of the network in Ironpython:
{
    "adjacency": {}, 
    "attributes": {
        "name": "Network"
    }, 
    "edge": {}, 
    "edge_attributes": {}, 
    "max_int_key": null, 
    "node": {}, 
    "node_attributes": {
        "x": 0.0, 
        "y": 0.0, 
        "z": 0.0
    }
}

Here is the print of the same network in python:
{
    "adjacency": {}, 
    "attributes": {
        "name": "Network"
    }, 
    "edge": {}, 
    "edge_attributes": {}, 
    "max_int_key": null, 
    "node": {}, 
    "node_attributes": {
        "x": 0.0, 
        "y": 0.0, 
        "z": 0.0
    }
}

salmamzfr, 2020-11-10:
Also tried **network.to_data** in python and loaded the data dictionary in Ironpython, which the correct dictionary, when it is loaded in Ironpython. Still get an empty network, while trying  **network.from_data**

brgcode, 2020-11-10:
not sure if this is the source of the problem, but on the CPython side you seem to be loading a `Mesh` and on the IronPython side a `Network`. could you check if this is correct? it would make sense that this results in the errors you describe...

salmamzfr, 2020-11-10:
**In case the mesh data structure:** I making the mesh in python using **Mesh.from_vertices_and_faces**, save it with **Mesh.to_json** and load it with **Mesh.from_json** and get the mentioned error.

salmamzfr, 2020-11-10:
In case of the network datastructure: I making the network in Python using **Network.from_nodes_and_edges**, save it with Network.to_json and load it in Ironpython with Network.from_json and get an empty network.

salmamzfr, 2020-11-10:
> not sure if this is the source of the problem, but on the CPython side you seem to be loading a `Mesh` and on the IronPython side a `Network`. could you check if this is correct? it would make sense that this results in the errors you describe...

Sorry! I pasted the wrong print, just corrected it!

brgcode, 2020-11-10:
could you send me the json file(s) that you have been using?

salmamzfr, 2020-11-10:
[Datastructures.zip](https://github.com/compas-dev/compas/files/5519348/Datastructures.zip)

brgcode, 2020-11-23:
```python
import os
from compas.datastructures import Network
from compas_rhino.artists import NetworkArtist

HERE = os.path.dirname(__file__)
FILE = os.path.join(HERE, 'Datastructures', 'pr_net')

net = Network.from_json(FILE)

artist = NetworkArtist(net)
artist.draw()
```

[image]

brgcode, 2020-11-23:
```python
import os
from compas.datastructures import Mesh
from compas_rhino.artists import MeshArtist

HERE = os.path.dirname(__file__)
FILE = os.path.join(HERE, 'Datastructures', 'pr_mesh')

mesh = Mesh.from_json(FILE)

artist = MeshArtist(mesh)
artist.draw()
```
[image]

brgcode, 2020-11-23:
@salmamzfr i can't really reproduce the problem. as you can see from the snippets and corresponding screenshots, for me these files just work...

salmamzfr, 2020-11-30:
Are you using two different compas versions? I am saving with 0.17.2-c422420c and drawing with 0.16.1

jf---, 2024-04-08:
> @salmamzfr i can't really reproduce the problem