compas-dev/compas #570: Installing COMPAS via Conda downgrades Matplotlib from 3.x to 2.x

tmsmendez, 2020-06-24:
**Describe the bug**
After a fresh Anaconda installation, I installed COMPAS on a recently made and activated environment. using Conda as described in the documentation. I noticed that the Matplotlib version was being downgraded from 3.x to 2.x as shown below in the screenshot.

**To Reproduce**
Steps to reproduce the behavior:
1. Context: Conda prompt
2. Sample script: conda install COMPAS -c conda-forge

**Expected behavior**
Compas installation without a downgrade of matplotlib version.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows virtual machine
 - Python version 3.7
 - Python package manager Conda

gonzalocasas, 2020-06-24:
The matplotlib version selected is related to the version of Python and the OS in use:
 - Windows, Python <=3.7 uses `matplotlib >=2.2,<3.1`
 - Windows, Python >=3.8, uses `matplotlib 3.1` and higher
 - MacOS & Linux, always uses `matplotlib >=2.2`

Bottom-line: create a new environment and use `python=3.8`

closed 2020-06-24 by gonzalocasas

tmsmendez, 2020-06-25:
Makes sense, thanks Gonzalo. 

Tomás