compas-dev/compas #572: method for sorting points by z (or x, or y) values

nizartaha, 2020-06-25:
# Feature Request

A nice little feature to have is to be able sort points by their z (or x, or y) coordinates. 

## Details

i needed it to know the upper corners of a bounding box in `compas.datastructures.mesh_bounding_box`

brgcode, 2020-06-25:
the bounding box is (or should be) a "structured" object.
does the following not work for you?
```python
box = mesh.bounding_box()

bottom = box[:4]
top = box[4:]

bottom_diagonal = bottom[0], bottom[2]
top_diagonal = top[0], top[2]

box_diagonal = bottom[0], top[2]
```

nizartaha, 2020-06-25:
It does work . Thank you Tom
Its beneficial to note it in the Documentation i guess. I couldn't tell from reading there. The top and bottom points under Examples seem duplicated.
https://compas-dev.github.io/main/api/generated/compas.datastructures.mesh_bounding_box.html#compas.datastructures.mesh_bounding_box

and maybe it is still beneficial to have a feature to sorting a collection of points by z or x or y. i dont have a specific case for it right now.

brgcode, 2020-06-25:
yes indeed, this should be documented. the example is also not great i guess...

jf---, 2024-04-22:
@tomvanmele can you consider closing this?

tomvanmele, 2024-04-23:
yes, the bounding box functions now return a box...

closed 2024-04-23 by tomvanmele