compas-dev/compas #574: Add: Mesh.delete_edge()

funkchaser, 2020-06-29:
# Feature Request

What is the intended way of deleting an edge in the compas `Mesh`? 
Similar classes (`Graph`, `Network`) have a `delete_edge()` method, but it seem to be missing in `Mesh`.
Thanks.

brgcode, 2020-07-06:
It depends on what you want the result to be. Edges are implicit objects in COMPAS meshes. So it is not as clear as in a network what the output is supposed to be. Should the adjacent faces be merged or should they both be deleted?

funkchaser, 2020-10-22:
By deleting an edge between two faces (e.g. if they are nearly co-planar) I want to merge both faces. This can lead to polygonal faces, which  - correct me if I'm wrong - is possible in compas Mesh. 
At the end of the day, I want the outlines of all resulting funnily-shaped faces. 
I could work with Network instead to delete edges, but - since there are no "faces" in Network - is there a way to "trace" the closed loops of edges?
Many thanks for any hint ;-)

brgcode, 2020-10-22:
there is a WIP function for this in the latest release `compas.datastructures.mesh_merge_faces`. it still seems to fail in some cases, but perhaps you could try it out and let me know?

funkchaser, 2020-11-09:
Thanks a lot! So far works fine with one catch: it seems that if two faces share more than one edge, upon merge some edges don't get removed -  I get"orphaned" edges. 

Example: Imagine a mesh consisting of 2x2 quad faces (4 faces in total). 
...... ......
| 1 | 3 |
...... ...... 
| 0 | 2 |
...... ......
I want to merge them into one big face. I go pair-wise merging the neighbouring faces: 
0 and 1 (--> new face 4), 
2 and 4 (--> new face 5, L-shaped), 
3 and 5 (--> new big face 6). 
Then I get an "orphaned" edge between the original faces 2 and 3 - the other shared edge was removed.
When I try calling `edge_faces` for this orphaned edge, I get a key error.

This looks like a bug, or some uncovered case?

brgcode, 2020-11-09:
that is a bug indeed. will have a look...

brgcode, 2020-11-23:
seems to work now...
```python
from compas.datastructures import Mesh
from compas_plotters import MeshPlotter

mesh = Mesh.from_vertices_and_faces([[0, 0, 0], [1, 0, 0], [1, 1, 0], [0, 1, 0]], [[0, 1, 2, 3]])
mesh = mesh.subdivide(scheme='quad')

mesh_merge_faces(mesh, [1, 2])
mesh_merge_faces(mesh, [3, 5])
mesh_merge_faces(mesh, [4, 6])

print(mesh.face_vertices(7))
print(mesh.face_halfedges(7))
for vertex in mesh.face_vertices(7):
    print(mesh.vertex_neighbors(vertex))

plotter = MeshPlotter(mesh, figsize=(8, 5))
plotter.draw_faces(text='key')
plotter.draw_edges()
plotter.draw_vertices(text='key', radius=0.03)
plotter.show()
```
```
[3, 5, 0, 4, 1, 6, 2, 7]
[(3, 5), (5, 0), (0, 4), (4, 1), (1, 6), (6, 2), (2, 7), (7, 3)]
[7, 5]
[3, 0]
[5, 4]
[0, 1]
[4, 6]
[1, 2]
[6, 7]
[3, 2]
```

[image]

jf---, 2024-04-22:
@tomvanmele curious what you think of this issue?

tomvanmele, 2024-04-23:
i think it is resolved

closed 2024-04-23 by tomvanmele