compas-dev/compas #575: compas_ghpython.draw_points() broken

yck011522, 2020-06-29:
I think the code for these drawing methods need some update

for example `pos = p['pos'] ` gives a bug on a new compas.Point

https://github.com/compas-dev/compas/blob/799536e87d4bb64a1c5812938f31a2d321e952aa/src/compas_ghpython/utilities/drawing.py#L70-L77

I don't have the latest HEAD at hand at the moment, but I believe the following code will not run in gh_python:
```
from compas_rhino.geometry import RhinoPoint
from compas_ghpython import draw_points
compas_points = []
compas_points  .append(RhinoPoint.from_geometry(point_input).to_compas())
output = draw_points(compas_points)
```

on my environment (c065d430021deed86fec09abee43ec7a87fa453a) it returns
```
Runtime error (TypeErrorException): not all arguments converted during string formatting

Traceback:
  line 169, in __getitem__, "C:\Users\leungp\AppData\Roaming\McNeel\Rhinoceros\6.0\Plug-ins\IronPython (814d908a-e25c-493d-97e9-ee3861957f49)\settings\lib\compas\geometry\_primitives\point.py"
  line 75, in draw_points, "C:\Users\leungp\AppData\Roaming\McNeel\Rhinoceros\6.0\Plug-ins\IronPython (814d908a-e25c-493d-97e9-ee3861957f49)\settings\lib\compas_ghpython\utilities\drawing.py"
  line 21, in script
```

yck011522, 2020-06-29:
As a hot fix , I changed line 75 to `pos = [p.x, p.y, p.z]` and the script will run. Perhaps there are better solutions, but I believe these functions need to be revised. 

On a bigger picture, I believe these Rhino/GH related functions are not CI "tested",  is there a way to use the new [Rhino Inside](https://www.rhino3d.com/inside) to do CI tests so we catch these problems before release?

brgcode, 2020-07-06:
not sure i understand the problem. the input for the drawing function is supposed to be a list of dicts with for each dict at least the key `'pos'`. the method `to_compas` of `RhinoPoint` only converts a point from Rhino to a point in COMPAS. therefore, i believe the snippet should be changed to the following before it can be expected to work:
```python
from compas_rhino.geometry import RhinoPoint
from compas_ghpython import draw_points
points = [{'pos': RhinoPoint.from_geometry(point_input).to_compas()}]
output = draw_points(points)
```

yck011522, 2020-07-07:
@brgcode 

Perhaps there was a misunderstand from my side about what the function does. Especially after reading the two word documentation: `Draw points.`

I do not know the history why this draw_points() function does not accept a list of points. But I believe the function signature is clear about performing `draw` on `point` and the `(s)` that indicate plural means it is probably a iteratable of points. I believe this function name is not the best if it intend to `draw a list of points in which each point coordinate is located in one of the key 'pos' in each dictionary`.

I suggest:
- reworking this part of the API to have a more human understandable function. Especially after the introduction of a bunch of Geometrical Classes. 
- improving documentation for these function from its current state.

gonzalocasas, 2020-07-07:
@brgcode shouldn't we have artists for point/frame/all others for `compas_ghpython` as well? that would solve this (recurrent) issue in the interpretation of this low-level API

yck011522, 2020-07-07:
@gonzalocasas  

When you say artist, do you mean it returns native Rhino objects? Or do you mean it handles the entire drawing routine (such as colour / opacity / layer)?

I find myself needing to do some Rhino native operations from time to time, and I find it convenient having converter for `compas geometry` to and from `rhino geometry`. 

At the same time it is also very relevant to have an artist that handles the drawing routine on CAD (Rhino in this case) canvas. I believe this was the intent behind the 'artist' concept. 

I think the two use case are slightly different and probably should both be supported. Particularly having a Rhino geometry does not always means painting it on the canvas. Hacking an `artist` to be a converter is weird and probably computationally expensive.

brgcode, 2020-07-07:
@gonzalocasas yes indeed. i will see what can be done about this in the short term.

@yck011522 this does not replace compas and/or rhino geometry objects but smoothens out some of the issues you have highlighted. the low-level drawing functions should not be used directly and have more intuitive artist-based equivalents like is currently already the case (to some extent) in `compas_rhino`...

brgcode, 2020-07-16:
this should be documented now. artists are in the works...

closed 2020-07-20 by yck011522