compas-dev/compas #577: function transfer script

xarthurx, 2020-07-06:
# Feature Request

Since along the development of compas/compas_fab/compas_xxx, there were constantly function name changing, location changing etc. that make the user's old script very hard to be adapted to the new release.

It would be really helpful to have a text parsing script that can automatically change whatever is changed in the new release to the new version.
 (similar to the one that parses python2 script to python3 script provided by the python3 development team)

Frankly speaking, it is really painful when trying to run the script I wrote just several months ago with the new release, and spent  time trying to figure out what is changed and how should I adapt it...

gonzalocasas, 2020-07-06:
We try to document changes in the changelog, so that you know what you need to change. If you don't want to spend time changing, then you can easily pin the compas version you depend on, and stick to it.

gonzalocasas, 2020-07-06:
A migration script in the style of python2to3 is not really practicable.

xarthurx, 2020-07-06:
@gonzalocasas  If only one version, that's fine... the problem in my case is I have different scripts for a different version of compas... 
Running different scripts by switching back-and-forth of compas/compas_fab is also not very practicable...

Why that hard? Can we just parse the file and replace the changed function? We don't need to implement sth fully like python2to3....

gonzalocasas, 2020-07-06:
Short answer is no, we can't.
The solution to this problem is really that you pin the version of COMPAS on your side.
Versioning is the mechanism that every other framework out there provides to give you control of how to adapt to a moving API.

> Running different scripts by switching back-and-forth of compas/compas_fab is also not very practicable...

I don't see why that would be ever needed, if you have different projects with different dependency sets, then you create different environments for each of then, pin compas/compas_fab versions and then each file will work with an expected API that does not change unless you actively change it.

> Why that hard? Can we just parse the file and replace the changed function?

It's not a trivial problem. Writing a find&replace script would only be poor and hacky attempt at tackling a problem that would require probably loading the abstract syntax tree of external code (code which we have no idea of its quality, code that could be written in any version of python, code that could be inside Grasshopper files, etc), and try to rewrite it to migrate to a new API. 

Just to illustrate the problem, how do you write a find&replace script that fixes the changed constructor of `Translation` if you could easily have python code that does `from compas.geometry import Translation as Tx; t = Tx(1, 2, 3)`?

Now imagine having to support this migrating scripts for every single feature we change in compas. We would not have time to get anything done.

The closest thing that is available in some frameworks are compatibility layers, eg. a separate package that you load and it makes a newer version behave like an older version. This would be possible to do, but we will only consider supporting this after we reach 1.0 and likely only between major version upgrades.

closed 2020-07-06 by gonzalocasas

xarthurx, 2020-07-06:
OK, now I see the problem...
Only thought about the function name, but didn't consider the function parameter and nickname issue...

Then hopefully the main functions can be fixed soon.