compas-dev/compas #578: Online documentation of previous compas release

yck011522, 2020-07-07:
# Feature Request

As a user of compas, compas_fab who pinned my repo to a certain version of compas / compas_fab, I want a online documentation of previous compas version release. so that it is easier to search for things.

At the moment: https://compas-dev.github.io/main/api.html does not have any reference to which version of compas it is related to.
## Details

**Describe the solution you'd like**
I would like to have a way of selecting other version of compas documentation when viewing it on the online doc.

It would be even more helpful to add a note to say since which version was each function / attribute / method added. This allows anyone to easily see which version to upgrade to, in order to have that function. 

**Describe alternatives you've considered**
Perhaps I can have a local documentation by building it. But having a online option would be very helpful.

tetov, 2020-07-15:
Duplicate of #341?

yck011522, 2020-07-15:
@tetov You are correct.

closed 2020-07-15 by yck011522