compas-dev/compas #585: How to get / set UserDictionary of a Rhino geometry object

yck011522, 2020-07-12:
# Feature Request
As a user of `compas_ghpython` and `compas_rhino` I would like to get an geometry by supplying a `guid`. And then reading or writing to the User Dictionary of that geometry. 

I could not find any documented functions about these. But I remember maybe forfin_ui did something like that.

I found this function [compas_rhino.utilities.get_object_attributes](https://compas-dev.github.io/main/api/generated/compas_rhino.utilities.get_object_attributes.html#compas_rhino.utilities.get_object_attributes) , but I couldn't figure out what it does. And despite this function looks like a `getter`, I couldn't not find a `setter`.

brgcode, 2020-07-16:
it is a getter and there is currently no setter, but i will remove the `get_` and combine them like in the rest of compas...

jf---, 2024-04-08:
@yck011522 good idea, any idea what the status is?

yck011522, 2024-04-09:
I believe @tomvanmele is taking care of this.