compas-dev/compas #586: Documentation missing for compas_rhino.geometry and .utilities

yck011522, 2020-07-12:
# Feature Request

As a user who need to depend on documentation, I want `compas_rhino` fuctions to be documented. Ideally with simple examples.

## Details

1. For example **all** the classes on this page have unclickable links: https://compas-dev.github.io/main/api/compas_rhino.geometry.html

2. Another example, 90% of all methods in this page do not have a single character of documentation: https://compas-dev.github.io/main/api/compas_rhino.utilities.html

--

_Despite I sound frustrated, don't get me wrong. I like how compas is evolving and what it aspire to become. But I have a feeling some parts of this library is not getting its fair share of love. I remember there were a series of opened issues about missing docs and examples months ago, not sure what happened to those. It is really hard for us (PhD student here) to use it for work like this, being able to IO with Rhino Geometry is fundamental to many users._

brgcode, 2020-07-16:
this should be solved now...

nortikin, 2020-07-20:
but it is not solved

brgcode, 2020-07-20:
the urls have changed

https://compas-dev.github.io/compas/api/compas_rhino.utilities.html
https://compas-dev.github.io/compas/api/compas_rhino.geometry.html

yck011522, 2020-07-20:
@nortikin 
Yes, I believe the permalink changed to the following:
https://compas-dev.github.io/compas/api/compas_rhino.geometry.html
https://compas-dev.github.io/compas/api/compas_rhino.utilities.html

closed 2020-07-20 by yck011522