compas-dev/compas #588: Feature Request: `compas` class to comply with some standard python JSONEncoder

yck011522, 2020-07-13:
Relating to discussion in #554.

It would be nice for all default `compas` class to comply with some standard python JSONEncoder. Such that any collections of `compas` objects, or user classes that contains `compas` objects can automatically and recursively serialize them all.

 I do not know how that works in Python. But I believe it is about subclassing the correct class and implement/override certain methods for de/serialization. In .net environment it is very easy to implement the ISerializable interface.

It would be nice to be able to just `json.dump(any_collection_of_compas_objects)`

_Originally posted by @yck011522 in https://github.com/compas-dev/compas/issues/554#issuecomment-657265719_

brgcode, 2020-08-07:
this should be working in the latest version. however, you have to use the `DataDecoder` and `DataEncoder` classes.
```python
import json
import compas
from compas.geometry import Point, Frame
from compas.datastructures import Mesh
from compas.utilities import DataEncoder, DataDecoder

point = Point(0, 0, 0)
frame = Frame.worldXY()
mesh = Mesh.from_obj(compas.get('faces.obj'))

data1 = [point, frame, mesh]

jsondata = json.dumps(data1, cls=DataEncoder)

data2 = json.loads(jsondata, cls=DataDecoder)

data1[0] == data2[0]
data1[1] == data2[1]
data1[2] == data2[2]  # note that this will not work because there is not comparison method implemented for meshes, but you get the point...
```

yck011522, 2020-08-07:
I'm not familiar with the `DataEncoder` classes but this looks amazing, I'll check it out soon.

gonzalocasas, 2020-09-21:
Is this now already solved? Should we close? or is it still dependant on releasing `0.17`?

brgcode, 2020-09-21:
no, it can be closed..

closed 2020-09-21 by gonzalocasas