compas-dev/compas #590: compas_cgal not yet public

yck011522, 2020-07-16:
# Feature Request

@brgcode Thanks for the presentation in the NCCR meeting today.
I'm interested in the [compas_cgal](https://github.com/BlockResearchGroup/compas_cgal) plugin. However, I believe it is not made public yet. 

I'm quite excited for the boolean operations. A quick question is, is this IronPython (GHPython in particularly) compatable or cpython only? If not can I call the boolean functions via proxy? Thanks

gonzalocasas, 2020-07-16:
it's the latter: CGAL-based code only works in cpython, but it can be used from GH using the proxy.

brgcode, 2020-07-16:
will make it public tomorrow. just have to update the docs...

brgcode, 2020-07-24:
`compas_cgal` is now public...

closed 2020-07-24 by brgcode