compas-dev/compas #593: intersection_line_line returns incorrect result when lines do not touch

yck011522, 2020-07-21:
**Describe the bug**

The following code contains two lines that do not intersect.
The result is not correct
```
from compas.geometry import Line, intersection_line_line, is_intersection_line_line
line1 = Line([0, 0, 0], [1, 0, 0])
line2 = Line([2, 0, 0], [2, 1, 0])
print (intersection_line_line(line1 , line2))
print (intersection_line_line(line2 , line1))
print (is_intersection_line_line(line1, line2))

>>>([2.0, 0.0, 0.0], [2.0, 0.0, 0.0])
>>>([2.0, 0.0, 0.0], [2.0, 0.0, 0.0])
>>>True
```

**Expected behavior**
The result of `intersection_line_line(line2 , line1)` should be unequal. According to the function description:

> list – XYZ coordinates of the two points marking the shortest distance between the lines. If the lines intersect, these two points are identical. If the lines are skewed and thus only have an apparent intersection, the two points are different. If the lines are parallel, the return value is [None, None].

I expect the Line object to have finite length and is not an infinite ray. According to Line doc:
> Line | A line is defined by two points.

From the doc of 'intersection_line_line', it said 'skewed' , I'm not sure if this is a professional terms, it is rather ambiguous to me. The way how I expect it to behave it that, when lines do not intersect, the two returned result should be at different locations. However that is not the case.

'is_intersection_line_line' is also returning 'True' for these two lines.
It appears the behavior of the two functions are treating the line as infinite rays. In that case perhaps the more acceptable naming should be called `infiniteline` or `ray`. Perhaps there should be a `Ray` object added to Geometry module to clarify between these two types, that would perhaps be useful in ray-casting type of functions in the future.

brgcode, 2020-07-21:
perhaps the docs could be clearer, but the result is correct.
a line is defined by two points, but it is not a segment between those points; it extends in either direction.
for segments there is a separate function: `intersection_segment_segment`

brgcode, 2020-07-21:
a ray, btw, has a starting point and is infinite in only one direction...

brgcode, 2020-07-21:
also, skew lines are three-dimensional lines that do not intersect and are not parallel. so this seems to be correct as well...

yck011522, 2020-07-21:
I think `intersection_segment_segment` do not exist yet. 
Only `intersection_segment_segment_xy` exist
Another note is `is_intersection_segment_segment` raise NotImplementedError

brgcode, 2020-07-21:
it exists, but it is missing from the docs. the predicate indeed raises an error...

yck011522, 2020-07-21:
This perhaps belongs to a separate Feature Request but I think the nomenclature of "Line" vs "Line Segment" is ambigous. Especially when "Line Segment" is not an class in itself. Alternatively, I think "Ray" vs "Line" is much less likely to be confused.

**Reason For**
Almost all CAD tools (say, Blender, Rhino) have a Line function that will draw a finite line. I think users generally expects the term "Line" as a finite line. 

**Reason Against**
That said, some math-vigorous sources consider line as infinite lines. [KhanAcademy](https://www.khanacademy.org/math/basic-geo/basic-geo-lines/lines-rays/a/lines-line-segments-and-rays-review) like to call those "lines, line segments, and rays". Perhaps this could be discussed in a separate thread whether compas should be more appealing to math-heads or cad-heads.

closed 2020-07-21 by yck011522