compas-dev/compas #600: Rhino paths to compas get lost after Rhino update

Achillx, 2020-07-30:
**Describe the bug**
Every time I update Rhino (**not** from Rhino 5 to Rhino 6) I need to run `python -m compas_rhino.install`

**To Reproduce**
Steps to reproduce the behavior:
1. Update rhino
2. Open EditPythonScript editor and try to `import compas`

**Expected behavior**
Not have to do it with every minor Rhino update

**Desktop (please complete the following information):**
 - OS: MacOS Mojave 10.14.6 (18G5033)
 - conda 2

**Additional context**
With every minor Rhino update, I do **not** have to reset the paths to my personal modules.

brgcode, 2020-08-24:
i will look into it but if the Rhino installer replaces the folder structure with every (major) update, i don't think there is much i can do about it...

Achillx, 2020-08-27:
It happened again today after installing the latest Rhino update.
I don't know if are aware of its existence, but I found it as I was looking around:
/Applications/Rhinoceros.app/Contents/Frameworks/RhCore.framework/Versions/A/Resources/ManagedPlugIns/RhinoDLR_Python.rhp/Lib/site.py

brgcode, 2020-08-28:
i am aware of the file and folders: that is where we symlink compas into. but that is more or less my point: if the entire folder tree gets replaced by a big update, i can't do anything about it...

gonzalocasas, 2020-08-28:
it would be interesting to see if after an update, the symlinks are present but broken, are if they are completely gone (due to whole folder being replaced)

brgcode, 2020-08-28:
i still have to do the update. will let you know...

Achillx, 2020-08-28:
I found an older Rhino6 version and I will also try

Achillx, 2020-08-28:
> i am aware of the file and folders: that is where we symlink compas into. but that is more or less my point: if the entire folder tree gets replaced by a big update, i can't do anything about it...

I don't think compas is doing what is described inside the `site.py` file.

Achillx, 2020-08-30:
Unfortunately, Rhino brings `/Applications/Rhinoceros.app/Contents/Frameworks/RhCore.framework/Versions/A/Resources/ManagedPlugIns/RhinoDLR_Python.rhp/Lib` folder back to its default state. I've also dropped some packages inside the `site-packages` folder to see if they are preserved, but no.

(_top left_ `site-packages`)
[image]

Achillx, 2020-08-30:
Adding symlinks inside `/Users/<username>/Library/Application Support/McNeel/Rhinoceros/6.0/scripts/` worked. This folder does not get overwritten after updates and also you don't have to restart Rhino when adding the symlinks.

Maybe that is an easy and quick fix?

brgcode, 2020-09-01:
can you register modules and packages there that are recognised by the RhinoPython interpreter?

Achillx, 2020-09-01:
Yes! Below you can see all symlinks that `compas_fab.rhino.install` adds to `/Applications/Rhinoceros.app/Contents/Frameworks/RhCore.framework/Versions/A/Resources/ManagedPlugIns/RhinoDLR_Python.rhp/Lib` moved to `/Users/<username>/Library/Application Support/McNeel/Rhinoceros/6.0/scripts/`.
Then when running `EditPythonScript` and typing
```
import compas
print(compas.__version__)
```
it works.
Also as I said before, you can add another folder or symlink while Rhino is running and the interpreter will see it.

[image]

brgcode, 2020-09-01:
okay cool i will check if this also works on windows...

brgcode, 2020-09-02:
@gonzalocasas that scripts folder is listed by default as one of the paths in "Module Search Paths" so it should work on Windows as well. what do you think? shall i make the switch?

gonzalocasas, 2020-09-02:
Yep 👍

brgcode, 2020-09-02:
done and tested on mac. @gonzalocasas culd you test on windows?

it will clean the old ironpython lib folder and then do the install/uninstall...

brgcode, 2020-09-02:
works on windows as well...

closed 2020-09-02 by brgcode