compas-dev/compas #606: ##[error]remote: Permission to compas-dev/compas.git denied to github-actions[bot].

KEERTHANAUDAY, 2020-08-24:
Hi !

I tried to submit a PR for docs #604 . My build fails at Deploy docs step with the error message,

```
##[error]remote: Permission to compas-dev/compas.git denied to github-actions[bot].
fatal: unable to access 'https://github.com/compas-dev/compas.git/': The requested URL returned error: 403
```

**To Reproduce**
Steps to reproduce the behavior:
1. Make docs change.
2. Make PR request

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - Windows 10
 - Python version 3.8.5
 - Python package manager - conda
- git version 2.27

**Additional context**
I tried upgrading my git version and passing a build twice and I received the same ever each time. I could not find any solution online.

gonzalocasas, 2020-09-01:
is this fixed?

brgcode, 2020-09-01:
no have had the same problem with other repos. seems that if you don't have write access the deploy thing doesn't work...

gonzalocasas, 2020-09-09:
As mentioned in one of the open PRs, this would be solved if we switch the `GITHUB_TOKEN` for `GH_PAT` (personal access token).

brgcode, 2020-10-19:
this should now be solved...

closed 2020-10-19 by brgcode