compas-dev/compas #610: Importing Grasshopper

beverlylytle, 2020-09-09:
**Describe the bug**
In a Grasshopper script, compas may not realize that it is running in Grasshopper.

**To Reproduce**
Steps to reproduce the behavior:
1. Run `compas_rhino.install` in a new compas environment.
2. Open a new Grasshopper canvas.

3. In a new GhPythonScript (with a Panel wired to `out`), run the following:
```
import compas
print(compas.GH)
```

**Expected behavior**
"True" should appear in the Panel.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Mac 10.15.5
 - Python version: IronPython
 - Python package manager: conda

**Additional context**
This problem was encounter with compas version 0.16.2-9f955a9f.

beverlylytle, 2020-09-09:
For anyone else who encounters this problem, a quick fix would be to add at the top of your script:
```
from compas_rhino import unload_modules
unload_modules('compas')

beverlylytle, 2020-09-22:
This appears to be out of date now, however I believe the issue is still the same with `compas.RHINO`.

beverlylytle, 2020-09-22:
No, it does not.  Closing the issue.

closed 2020-09-22 by beverlylytle

brgcode, 2020-09-22:
i have removed `compas.GH` because we don't actually use it anywhere...