compas-dev/compas #615: Backwards incompatible serialization

gonzalocasas, 2020-09-14:
**Describe the bug**
The `DataEncoder` class serializes data objects exposing the internal module name instead of the public one. This effectively break backwards compatibility because prior versions will fail to decode, and will potentially break down the road as well since it is leaking internal implementation details.

This is probably happening since the removal of the hard-coded `__module__` attribute.

**To Reproduce**
Steps to reproduce the behavior:
```python
>>> from compas.utilities import DataEncoder
>>> from compas.geometry import Frame
>>> import json
>>> json.dumps(Frame.worldXY(), cls=DataEncoder)
'{"dtype": "compas.geometry.primitives.frame/Frame", "value": {"point": [0.0, 0.0, 0.0], "xaxis": [1.0, 0.0, 0.0], "yaxis": [0.0, 1.0, 0.0]}}'
```

**Expected behavior**
 - Got `"dtype": "compas.geometry.primitives.frame/Frame"`
 - Expected: `"dtype": "compas.geometry/Frame"`.

**Desktop (please complete the following information):**
 - `compas 0.16.2-4831754c`

brgcode, 2020-09-14:
perhaps we could add some kind of module attribute to the classes that does not interfere with the internal discovery mechanisms of python itself, but that gets picked up by the encoder?

brgcode, 2020-09-14:
or we could use `__getstate__` and `__setstate__`. these will automatically affect the pickling process and we could just have the decoder and encoder classes pick it up as well?

gonzalocasas, 2020-09-14:
> or we could use __getstate__ and __setstate__. these will automatically affect the pickling process and we could just have the decoder and encoder classes pick it up as well?

that sounds interesting.

brgcode, 2020-09-14:
shall i try something?

gonzalocasas, 2020-09-15:
workaround available since https://github.com/compas-dev/compas/commit/c05d337878636b79237420077ca6a8f53d01a39e

@brgcode should we close? or want to keep open until a more definitive solution is in place?

brgcode, 2020-09-15:
no let's close

closed 2020-09-15 by brgcode