compas-dev/compas #620: smarter path detection of rhino installation folder

tkmmark, 2020-09-22:
For those with multiple version of Rhinos installed, or Rhinos installed in odd directories, the current hard-coded _get_ironpython_lib_path_OS does not always work

For instance... my Rhino 6 is installed inside Rhinoceros2.app folder, not Rhinoceros.app.    As such, I must edit line 79 and line 124 in the __init__.py of compas_rhino to get the compas_rhino installation working.

brgcode, 2020-09-24:
did Rhino change that name or did you? :) because i have `Rhinoceros.app` and `RhinoWIP.app`...

tkmmark, 2020-10-30:
Not sure how it happened... Probably not... I would have preferred to rename it according to its version number.

jf---, 2024-04-08:
@tomvanmele old let's keep weeding stale issues

closed 2024-04-08 by tomvanmele