compas-dev/compas #625: delete_node not deleting out edge.

yck011522, 2020-10-05:
When a two edges exist between two nodes, in a Network object.
Such as a-b and b-a. When one node is deleted, only one of the two edges are removed.

**To Reproduce**
```
from compas.datastructures import Network

n = Network()
n.add_node('a')
n.add_node('b')
n.add_edge('a' , 'b')
n.add_edge('b' , 'a')

print ('--- node a and b  with bi directional edge---')
print ('nodes' , list(n.nodes()))
print ('edges' , list(n.edges()))
print ('neighbors of a : ' , n.neighbors('a'))
print ('neighbors_out of a : ' , n.neighbors_out('a'))

n.delete_node('b')

print ('--- after removing b ---')
print ('nodes' , list(n.nodes()))
print ('edges' , list(n.edges()))
print ('neighbors of a : ' , n.neighbors('a'))
print ('neighbors_out of a : ' , n.neighbors_out('a'))
```
**Output**
```
--- node a and b  with bi directional edge---
nodes ['a', 'b']
edges [('a', 'b'), ('b', 'a')]
neighbors of a :  ['b']
neighbors_out of a :  ['b']
--- after removing b ---
nodes ['a']
edges [('a', 'b')]
neighbors of a :  []
neighbors_out of a :  ['b']
```

Note that edge `('a', 'b')` still exist and 'b' is still listed as a neighbors_out of a.

**Expected behavior**
I expect both edges to be deleted.

**Desktop (please complete the following information):**
- compas 0.16.6 (pip)
- python 3.7

brgcode, 2020-10-28:
@yck011522 can this be closed?

yck011522, 2020-10-28:
Hi,

Just updated to 0.16.9 and run the same code again. This doesn't look right.
I think ` n.neighbors_out('a')` is a valid query and should not result in a key error.

[image]

yck011522, 2020-10-28:
Did some more test:

[image]

Note that line 8 did not throw an error but line 26 does. Seems like the delete_node didn't clean everything up properly.

brgcode, 2020-11-23:
@yck011522 should be solved since 8cf5933a
please check and let me know...

yck011522, 2020-12-07:
Yep, I haven't checked the code but the behaviour seems good now.

closed 2020-12-07 by yck011522