compas-dev/compas #630: compas predicate support for compas geometry

tkmmark, 2020-10-10:
It would be great if the various compas.geometry methods (e.g. predicates, distance, etc) that take geometric information as arguments (e.g point, circles etc) can fully support the use of compas geometry as inputs.

For instance, in 'is_point_in_circle_xy', consider adopting 359-360 (or something more pythonic and elegant), so that the method can take both compas.geometry.Circle as inputs or a (point, radius) tuple.

[image]

brgcode, 2020-10-11:
i think this is only the case for circles. afaik, all other objects can be indexed correctly. so perhaps we should just update the definition of a circle, because in many cases it is unnecessarily verbose...

tkmmark, 2020-10-11:
Will not disagree with simplifying the circle too, or at least simplifying the constructor so that one can instantiate with a point, and a radius (assumes xy plane by default).  This is something I already do in my extended Circle class. 
 As for updating the definition, I always thought the plane was included because there are designs for more extended functionalities in 3-D, or transformed spaces in the future ... so you will be the best judge on that.

jf---, 2024-04-08:
hi @tomvanmele what about this one? i like the `cgal` predicate thing, this is just about weeding out the issue tracker bit by bit

closed 2024-04-08 by tomvanmele