compas-dev/compas #633: regarding 'proper processing' of edgedata from older jsons

tkmmark, 2020-10-12:
Assuming that the intention of the 'else' portion of the data.setter in halfedge is to enable the loading of jsons generated using the deprecated scheme (before early September)...  shouldn't edge keys be converted from the older format to the newer format inside the datasetter so that the edge-based methods in the newer compas can actually make use of the saved edge attributes (imported from the json with the older data format)?

so... something like this:

            for uv, attr in iter(edgedata.items()):
                uv = sorted(literal_eval(uv))
                uv = '-'.join([str(uv[0]), str(uv[1])])
                self.edgedata[uv] = attr or {}
                
otherwise... the edge attributes are out of reach by the newer compas methods.

tkmmark, 2020-10-12:
and certainly, this example above is not optimised since edge attributes will be input twice (ie. uv and vu)

brgcode, 2020-10-17:
thanks for pointing this out. it was indeed an issue! should be solved in dc3b06f8

brgcode, 2020-10-28:
@tkmmark can this be closed?

tkmmark, 2020-10-28:
Didn’t pull from the latest Master yet (too scared of what new surprises it may hold :P), but this commit looks okay, so sure.

closed 2020-10-28 by tkmmark