compas-dev/compas #636: edge_attributes() key error

tkmmark, 2020-10-16:
The edge key conversion from (int, int) to str(int)+'-'+str(int) in edge_attributes causes problems, since it uses edge_attribute and edge_attribute already take cares of the conversion.  So when one uses edge_attributes now, one receives an error as the key is already converted to string when edge_attribute is called inside by edge_attributes, and therefore the key cannot be unpacked into u, v

tkmmark, 2020-10-16:
_Would suggest removing_ 

key = "-".join(map(str, sorted(edge)))

_and replacing the first line with_

u, v = key = edge

_and placing inside 'if not names' the following_

key = "-".join(map(str, sorted(edge)))

brgcode, 2020-10-17:
this should also be fixed since dc3b06f8

brgcode, 2020-10-28:
@tkmmark can this be closed?

closed 2020-10-28 by tkmmark