compas-dev/compas #647: Deepcopy Network object fail

yck011522, 2020-10-28:
**Describe the bug**
I cannot run copy.deepdopy on Network object.

**To Reproduce**

```
from compas.datastructures import Network

from copy import deepcopy
network = Network()
network2 = deepcopy(network)
print (network3.to_data())
```

**Error:** 'Network' object has no attribute 'attributes'
..... line 4, in <module>  network2 = deepcopy(network)

**Expected behavior**
I expect deepcopy to work. It is a pretty common code pattern

**Desktop (please complete the following information):**
 - OS: [e.g. Windows 10]
 - Python version [e.g. 3.7.7 and IPY] both fail
 - Python package manager [e.g. pip, conda]

This also happens to other classes:

Exception has occurred: AttributeError
'Graph' object has no attribute 'attributes'
Exception has occurred: AttributeError
'Mesh' object has no attribute 'attributes'

brgcode, 2020-10-28:
should be solved since e5b1e234
please check...

yck011522, 2020-10-28:
Thanks for the quick reply. 

dumb problem: I pulled the compas source and tried pip install it from source, but pip wont let me because compas_fab requires "compas<0.17,>=0.16.9" .. How should I manage this, a.k.a, what do i do...?

yck011522, 2020-10-28:
Hum. I'm trying the `--no-dependencies` route, lets see if it breaks other things...

yck011522, 2020-10-28:
Okay, I confirm that e5b1e23 do not have the problem I mentioned

closed 2020-10-28 by yck011522