compas-dev/compas #648: [Pybullet backend] Attached collision object does not update after robot configuration is updated

yijiangh, 2020-10-28:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
compas_fab version `0.12.0`:

MWE:
```python
import time

import compas_fab
from compas_fab.backends.pybullet import PyBulletClient
from compas_fab.robots import AttachedCollisionMesh
from compas_fab.robots import CollisionMesh
from compas_fab.robots import Configuration

from compas.datastructures import Mesh

with PyBulletClient() as client:
    urdf_filepath = compas_fab.get('universal_robot/ur_description/urdf/ur5.urdf')
    robot = client.load_robot(urdf_filepath)

    mesh = Mesh.from_stl(compas_fab.get('planning_scene/cone.stl'))
    cm = CollisionMesh(mesh, 'tip')
    acm = AttachedCollisionMesh(cm, 'ee_link')
    client.add_attached_collision_mesh(acm, {'mass': 1, 'robot': robot})
    input('Attached!')

    # update robot conf
    conf = Configuration.from_revolute_values([1.0,0,0, 0,0,0])
    client.set_robot_configuration(robot, conf)

    time.sleep(1)
    client.step_simulation()
    time.sleep(1)
    input('update conf!')

    client.remove_attached_collision_mesh('tip')
    input('Removed!')

    time.sleep(1)
```

At the `Attached` state:
[image]

At the `'update conf!` state:
[image]

At the `removed!` state:
[image]

**Expected behavior**
The attached collision object should stay attached to the robot's flange after the configuration update.

**Desktop (please complete the following information):**
 - OS: Windows
 - Python version: 3.7
 - Python package manager: pip

yck011522, 2020-10-29:
I think this should go to the compas_fab issue tracker?

yijiangh, 2020-10-29:
@yck011522 You are absolutely right! I had too many tabs open in my browser... Sorry for the confusion.

closed 2020-10-29 by yijiangh