compas-dev/compas #649: Performance issue for create object in RhinoPython

yck011522, 2020-10-29:
**Describe the bug**
Creating object is very slow in RhinoPython. There seems to be a memory leak.
compas.version = '0.17.1' pulled from source e5b1e23484f6951c1c2013dfa58ae16349ddfdd4

**To Reproduce**
This function test3() creates 3 lines, the test runs this 100 times:
```
from compas.geometry import Line

def test3():
    line1 = Line([-100, 0, 0], [100, 0, 0])
    line2 = Line([0, -100, 0], [0, 100, 0])
    line3 = Line([0, -100, 10], [0, 100, 10])

if __name__ == "__main__":
    import datetime

    first_time = datetime.datetime.now()
    for i in range(100):
        test3()
    later_time = datetime.datetime.now()
    difference = later_time - first_time
    print("Time took: %s" % difference)
```
**Running this is cPython 3.7** 
Time took: 0:00:00.004987

**Running this in Rhino (fresh after ResetScriptEngine)**
Time took: 0:00:00.051000
**Subsequent runs:**
Time took: 0:00:00.319000
Time took: 0:00:00.490000
Time took: 0:00:01.155000
Time took: 0:00:01.173000
Time took: 0:00:01.154000
Time took: 0:00:02.093000
Time took: 0:00:02.136000
Time took: 0:00:02.133000
Time took: 0:00:02.648000
you get the idea, it keeps going up... until Rhino never returns (aka crash)

yck011522, 2020-11-03:
ping ping.
Perhaps just to reiterate, This is a breaking bug that crashes Rhino.

gonzalocasas, 2020-11-03:
The problem (again) comes down to the `abc` module. The `Base` class of compas inherits from `ABCMeta` which apparently has a very bad memory leak on IronPython's version of Rhino. I will send a PR getting rid of it. I'm still not sure what are the side effects of this change.

commit 662659f9 (2020-11-04): Merge pull request #651 from compas-dev/no-abc-is-left-inside

closes #649: remove usage of ABCMeta to prevent performance issues on…
commit 9e292750 (2020-11-04): closes #649: remove usage of ABCMeta to prevent performance issues on IronPython
closed 2020-11-04 by brgcode