compas-dev/compas #655: Docs Suggestion: using DataDecoder, DataEncoder with custom user classes

yck011522, 2020-11-06:
# Feature Request

I figured that I could encode and decode my own class objects with the same `DataDecoder` , `DataEncoder` system. As long as I implemented the following functions:

- `def to_data(self):`
- `@property    def data(self):`
- `@classmethod def from_data(cls,data):`

This works also if my objects is recursively placed within a compas.datastructure type objects, such as a `Network`. This function is fantastic (kudos to all devs who made this happen) and I think more people should be aware of it. Perhaps a simple page of documentation / example code can be added either within the pages of `DataEncoder` and `DataDecoder`. or in `Tutorials`.

jf---, 2024-04-08:
@yck011522 is this still relevant?

tomvanmele, 2024-04-08:
even if it is, it is most certainly a duplicate...

closed 2024-04-08 by tomvanmele