compas-dev/compas #656: Doc Suggest: Can we add description how these update functions changes the internal state of objects

yck011522, 2020-11-06:
As titled, I was trying to understand what this update function does and realize that the word 'update' doesn't mean much to someone trying to learn. Especially this function is a few calls deep before the actual line of code to change things appear. 

Perhaps it is better to add a line to explain what internal states are changed after the function call. Better if we can have a line that describes when this call is necessary.

https://github.com/compas-dev/compas/blob/c422420c4cb4b2f1e41f475e938d86d3ba71fb9b/src/compas/robots/base_artist/_artist.py#L225-L226

yck011522, 2020-11-07:
Same for the `update_tool` function. 

https://github.com/compas-dev/compas/blob/75d009d9cdbc2309d606a39a8e3de3c6846535d3/src/compas/robots/base_artist/_artist.py#L259-L277

I'm not sure what exactly does the description want to say: 

- Is there an internal state in the tool object that holds the transformed position of the joints, that got updated?
- Is the artist going to re-paint the geometry in the front end UI?
- What will happen if I supply a `joint_state` input, what will happen if I don't?
- When will anyone need to supply `transformation`, isn't the tool attached to the ee of the robot? Or is this transform going to override the ee_frame? Or does it transform again after ee transform? it says:

> The (absolute) transformation to apply to the entire tool's geometry.
   
- Do I need to call `update_tool `if I have already called `update `?

tomvanmele, 2024-04-23:
gonna close this since `compas.robots` doesn't exist anymore...

closed 2024-04-23 by tomvanmele