compas-dev/compas #660: api reference showing draw_collection functions that have been removed

mattiskoh, 2020-11-10:
**Describe the bug**
A clear and concise description of what the bug is.
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Context [e.g. ST3, Rhino, Blender, ...]
2. Sample script
3. Sample data
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Python version [e.g. 2.7]
 - Python package manager [e.g. macports, pip, conda]

**Additional context**
Add any other context about the problem here.

brgcode, 2020-11-23:
i think this is no longer the case. could check and close if solved? thanks...

gonzalocasas, 2020-11-30:
nope, they are still showing up.

brgcode, 2020-11-30:
i guess this is because the default configuration of invoke docs changed...

gonzalocasas, 2020-11-30:
@brgcode you mean, to not rebuild? but this is done on CI, so, I don't think there's any cached data/leftover from previous builds.

brgcode, 2020-11-30:
@Licini could you have a look at this...

beverlylytle, 2020-12-08:
`draw_collection` has been removed for `PolylineArtist` and `LineArtist` in compas_rhino, but not in compas_ghpython.  So the API reference is accurate.  I guess the question is, which `draw_collection` methods should stay and which should go?

brgcode, 2020-12-08:
i think they should all go :)

beverlylytle, 2020-12-08:
Even the recently added `BoxArtist.draw_collection`?

brgcode, 2020-12-08:
yup. unless you object :)

beverlylytle, 2020-12-08:
Not at all.

PR #693 (2020-12-08): remove all draw_collection methods
Closes #660.  

I simply deleted a lot of code.  But it could be argued that it would be cleaner to leave the methods in the base artists raising not implemented errors (and thus could be considered a non breaking change), but not implemented errors in my mind have an implicit "yet" attached to them.  Thoughts?

### What type of change is this?

- [ ] Bug fix in a **backwards-compatible** manner.
- [ ] New feature in a **backwards-compatible** manner.
- [x] Breaking change: bug fix or new feature that involve incompatible API changes.
- [ ] Other (e.g. doc up [... cut]
closed 2020-12-08 by brgcode