compas-dev/compas #661: unify meshplotter within geometryplotter (artist) framework

tkmmark, 2020-11-10:
Designers often need to simultaneously visualise stand-alone geometries with COMPAS datastructures like Meshes or Networks (e.g. seeing the results of trimesh_pull_points).   However, there does not seem to be a way, to the best of my knowledge, to do this easily with COMPAS right now without having to resort to hacking something together with the draw methods in **compas_plotter.core** and directly accessing the matplotlib axes embedded in the plotter/artist.  Would be great to bring the older plotter setup under the newer geometry artist framework, so that the workflow for developers can be simplified into something closer to...

plotter = GeometryPlotter()
plotter.add(mesh)
plotter.add(lines)
plotters.show()

tkmmark, 2020-11-10:
As a quick fix, even some mesh_plot_instance.to_geometry_artists() would be helpful.

brgcode, 2020-11-10:
it is indeed the goal to allow all types of COMPAS objects to be added to plots in the style supported by the geometry plotter. in the meantime, can you not use the methods `draw_points`, `draw_lines`, `draw_...` of the `MeshPlotter`?

tkmmark, 2020-11-10:
yes, I can, which brings me to a suggestion I have forgotten to include in my last comment: "'it would also be helpful, as a quick interim fix, to introduce some methods like `geom_artist.to_plotter_dict()`"

tkmmark, 2020-11-10:
The preference for geometry plotter is due to its ability to handle style in a model-scale-invariant way... (see figure)
[image]
top row: mesh plotter
bottom row: geom plotter
left col: scale factor 1
right col: scale factor 3

jf---, 2024-04-22:
@tomvanmele || @gonzalocasas bump

gonzalocasas, 2024-04-22:
I think this should be closed, the plotters don't exist anymore

gonzalocasas, 2024-04-22:
re-open if i'm wrong though :)

closed 2024-04-22 by gonzalocasas