compas-dev/compas #662: draw_points argument in PolylineArtist not functional

tkmmark, 2020-11-10:
Consider making the following insertions (as highlighted):

[image]

For copy-and-paste-convenience, the code has been reproduced here as text:

self._points_artists = []

if self._draw_points:
	for point in self.polyline:
		self._points_artists.append(self.plotter.add(point))

Setup is based on LineArtist... and tested, as shown below:

[image]

jf---, 2024-04-22:
@tomvanmele || @gonzalocasas, any thoughts on this contribution?

gonzalocasas, 2024-04-22:
make sense, I haven't checked if this is still the case with the new 2.0 release and the scene stuff. @tkmmark would you mind creating a PR for this?

tomvanmele, 2024-04-23:
this is related to the plotter, which doesn't exist anymore...

closed 2024-04-23 by tomvanmele