compas-dev/compas #669: RobotModelArtist.update() fails to update the position of PRISMATIC,REVOLUTE joints.

tristanryerparke, 2020-11-15:
This issue is appearing in Blender and Rhino on macOS.
Blender 2.90.0, Rhinoceros 6(Atom).
When trying to display positions of the first axis of a five axis cartesian robot with these simple scripts:
Rhinoceros:
```
import compas
from compas.robots import RobotModel, Joint
from compas_rhino.artists import RobotModelArtist
from compas.geometry import Point, Vector,Frame
from compas.datastructures import Mesh
compas.PRECISION = '12f'
robot = RobotModel("robot", links=[], joints=[])
meshFolder = "/Users/tristanryerparke/Documents/Python/PM_Python/machineMeshes/"
#Define first link
mesh_static = Mesh.from_stl(meshFolder + "static.stl")
static_axis = robot.add_link("static",mesh_static,(0.2, 0.5, 0.6))
#Define second link
mesh_y_axis = Mesh.from_stl(meshFolder + "y_axis.stl")
y_axis = robot.add_link("yaxis",mesh_y_axis,(0.2, 0.5, 0.6))
#Define Vector, Plane and Limits for first joint
y_dir = Vector(0, 0, 1)
home_point = Point(0,0,0)
home_frame = Frame.worldXY()
#When the y_limits is set to [-1,3] and the Joint is set to PRISMATIC,
#[-1,3], the y_axis apears -1 units lower, so I know the units are correct.
y_limits = [0,-3] 
#When switching the joint type to CONTINUOUS, 
#updating the artist with y_joint values will move the mesh correctly. However,
#when switching the joint to PRISMATIC or REVOLUTE, the value does not affect the position of the drawn mesh.
robot.add_joint("y_joint",Joint.PRISMATIC,static_axis,y_axis,home_frame,y_dir,y_limits)
artist = RobotModelArtist(robot,'PM')
artist.update({"y_joint":-0.5})
artist.clear_layer()
artist.draw_visual()
```
Blender:
```
import compas
from compas.robots import RobotModel, Joint
from compas_rhino.artists import RobotModelArtist
from compas.geometry import Point, Vector,Frame
from compas.datastructures import Mesh
compas.PRECISION = '12f'
robot = RobotModel("robot", links=[], joints=[])
meshFolder = "/Users/tristanryerparke/Documents/Python/PM_Python/machineMeshes/"
#Define first link
mesh_static = Mesh.from_stl(meshFolder + "static.stl")
static_axis = robot.add_link("static",mesh_static,(0.2, 0.5, 0.6))
#Define second link
mesh_y_axis = Mesh.from_stl(meshFolder + "y_axis.stl")
y_axis = robot.add_link("yaxis",mesh_y_axis,(0.2, 0.5, 0.6))
#Define Vector, Plane and Limits for first joint
y_dir = Vector(0, 0, 1)
home_point = Point(0,0,0)
home_frame = Frame.worldXY()
#When the y_limits is set to [-1,3] and the Joint is set to PRISMATIC,
#[-1,3], the y_axis apears -1 units lower, so I know the units are correct.
y_limits = [0,-3]
#When switching the joint type to REVOLUTE,
#updating the artist with y_joint values will move the mesh correctly. However,
#when switching the joint to PRISMATIC, the value does not affect the position of the drawn mesh.
robot.add_joint("y_joint",Joint.CONTINUOUS,static_axis,y_axis,home_frame,y_dir,y_limits)
artist = RobotModelArtist(robot,'PM')
artist.update({"y_joint":-0.5})
artist.clear_layer()
artist.draw_visual()
```
I found it strange that the issue persists across blender and rhino. 
Cheers,
T

beverlylytle, 2020-11-16:
Hi, thanks for bringing up this issue!  I think the problem might be how the argument `limit` of `add_joint` is used.  There's an assumption that the limit is given in the order `[<lower_limit>, <upper_limit>]`.  So if you set `y_limit = [-3, 0]` the prismatic joint values can be updated to `-0.5`.  I suppose we could do one of two things to address this: 1. state this assumption explicitly in the documentation, or 2. change the internals of `add_joint` to infer which of the two elements is the lower limit.  @gonzalocasas what do you think?

gonzalocasas, 2020-12-01:
@beverlylytle: spot on! the problem is that limits are reversed, the first one needs to be the lower bound. 

The documentation states this and also indicates these limits only apply to prismatic and revolute joint, so it makes sense that using continuous joints seems to work, because the limit is not applied.

[image]

@tristanryerparke could you please confirm if this works for you?

gonzalocasas, 2021-01-04:
@tristanryerparke ping

closed 2021-02-01 by gonzalocasas