compas-dev/compas #670: Add Urdf writers

gonzalocasas, 2020-11-16:
Currently, the URDF support can only read/parse urdf files, but cannot write them back to disk if they are modified.
Adding urdf writer support would help in other areas, such as mix&match different robot clients in compas_fab

gonzalocasas, 2020-11-16:
See https://github.com/compas-dev/compas_fab/issues/232 for more details

brgcode, 2020-11-23:
could we upgrade our parser with something like this?
https://dustinoprea.com/2019/01/22/python-parsing-xml-and-retaining-the-comments/

brgcode, 2020-11-23:
or use BeautifulSoup instead?

beverlylytle, 2020-11-26:
> could we upgrade our parser with something like this?

I looked into this, but I was unsuccessful.  There were a few obstructions.  One is that many URDFs have comments outside the root element of the xml, which, pre python 3.8, xml.etree strongly objects to.  When I tried to snake my way around that, I discovered that there were many changes made within xml.etree over the various versions of python 3, and it just got too complicated.

brgcode, 2020-11-26:
can you post a minimal xml string that has these items we can currently not deal with?

beverlylytle, 2020-11-26:
```"<Tests></Tests>"```
The current parser can handle this, but it will strip out all the comments.  I can't figure out a way to preserve them.

PR #681 (2020-12-01): Urdf writer
Closes #670. 

### What type of change is this?

- [ ] Bug fix in a **backwards-compatible** manner.
- [X] New feature in a **backwards-compatible** manner.
- [ ] Breaking change: bug fix or new feature that involve incompatible API changes.
- [ ] Other (e.g. doc update, configuration, etc)

### Checklist

_Put an `x` in the boxes that apply. You can also fill these out after creating the PR. If you're unsure about any of them, don't hesitate to ask. We're here to help! This is simply a reminder of what we are going to look for before merging your code._

- [X] I added a line to th [... cut]
closed 2020-12-01 by beverlylytle