compas-dev/compas #671: multiple things re: meshplotter's update_vertices and general collections management

tkmmark, 2020-11-16:
Currently, update_vertices just generically redraw the circles associated with each vertex of a mesh... not the vertices that, say, a user might have specified when draw_vertices was initially called.  Ideally, update_vertices should act according to a meshplotter instance's own history of draw_vertices().  It follows from this that things like text is also ignored in the update.

I suspect this will require the plotter to store the list of dictionaries associated with each draw_element(), not only the axes's collection, so that the update_element() method will know what else requires updating...

On that note, the meshplotter class currently only store the elementcollection (as matplotlib collections) associated with the final draw_element(), which also mean there will lingering geometries when clear() is invoked if a user had called draw_element() (for the same element type, e.g. vertex) multiple times.   (A user may use draw_vertices multiple time in order to draw different type of vertices with different settings)

brgcode, 2020-11-17:
the current implementation is indeed limited and not very flexible. afaik it can only be used in one specific way. however, this will go away when rebasing onto the new plotter system...

tomvanmele, 2024-04-23:
plotter doesn't exist anymore

closed 2024-04-23 by tomvanmele