compas-dev/compas #675: mesh.vertices_on_boundary() for N-gon mesh

matteo-pacher, 2020-11-24:
**Describe the bug**
The function `mesh.vertices_on_boundary()` never arrives to the exit condition of the while loop, appending infinitely to the list `vertices`, and eventually causing a `Runtime error (OutOfMemoryException): Array dimensions exceeded supported range`.
The definition of the `start` key might be the problem.

**To Reproduce**
Script to reproduce the behavior in Python 3.7:
```
mesh_json = {
    "compas": "0.17.2",
    "datatype": "compas.datastructures/Mesh",
    "data": {
        "attributes": {
            "name": "Mesh"
        },
        "dva": {
            "x": 0.0,
            "y": 0.0,
            "z": 0.0
        },
        "dea": {},
        "dfa": {},
        "vertex": {
            "0": {
                "x": 17.56081771850586,
                "y": 4.622284412384033,
                "z": 1.2479242086410522
            },
            "1": {
                "x": 18.21266746520996,
                "y": 5.342650413513184,
                "z": 2.371530055999756
            },
            "2": {
                "x": 20.60313606262207,
                "y": 5.956735610961914,
                "z": 2.9836647510528564
            },
            "3": {
                "x": 17.613462448120117,
                "y": 5.7198333740234375,
                "z": 2.9479243755340576
            },
            "4": {
                "x": 19.751209259033203,
                "y": 7.413698196411133,
                "z": 2.355659008026123
            },
            "5": {
                "x": 20.739913940429688,
                "y": 6.193633556365967,
                "z": 2.9836647510528564
            },
            "6": {
                "x": 17.75023651123047,
                "y": 5.956733703613281,
                "z": 2.9479243755340576
            },
            "7": {
                "x": 21.414566040039062,
                "y": 6.00639533996582,
                "z": 2.5479743480682373
            },
            "8": {
                "x": 17.33991241455078,
                "y": 5.719833850860596,
                "z": 2.9479243755340576
            },
            "9": {
                "x": 21.61810302734375,
                "y": 7.367147445678711,
[... 298 lines cut ...]
            "23": {}
        },
        "edgedata": {},
        "max_vertex": 35,
        "max_face": 23
    }
}

from compas.datastructures import Mesh
built_mesh = Mesh.from_data(mesh_json)
built_mesh.edges_on_boundary()
```

**Expected behavior**
Returning the boundary edges of the mesh.

**Desktop:**
 - OS: Windows 10
 - Python version: 3.7
 - Python package manager: conda
 - compas version: 0.17.2

**Additional context**
Are N-gon meshes generally supported/accepted in compas?
What other functions could lead to problems for N-gon meshes?

brgcode, 2020-11-24:
[image]

before looking in to this further, can you confirm that this is the correct mesh?

brgcode, 2020-11-24:
there is a vertex which belongs to 2 boundaries and this causes the problem...

matteo-pacher, 2020-11-24:
The mesh is correct. Mesh.is_valid() didn't throw any errors

brgcode, 2020-11-24:
i have pushed an update of the boundary functions to a PR. once merged it should solve your problem...

matteo-pacher, 2020-11-25:
I have been testing the fix, and it seems to solve the problem.
Thank you Tom!

matteo-pacher, 2020-11-25:
I found a case where the fix does not work:

```
mesh_json = {
    "compas": "0.17.2", 
    "data": {
        "attributes": {
            "name": "Mesh"
        }, 
        "dea": {}, 
        "dfa": {}, 
        "dva": {
            "x": 0.0, 
            "y": 0.0, 
            "z": 0.0
        }, 
        "edgedata": {}, 
        "face": {
            "0": [
                34, 
                16, 
                20, 
                12, 
                11
            ], 
            "1": [
                20, 
                21, 
                7, 
                27, 
                12
            ], 
            "2": [
                27, 
                7, 
                19, 
                30, 
                15
            ], 
            "3": [
                10, 
                1, 
                3, 
                8, 
                18
            ], 
            "4": [
                35, 
                23, 
                18, 
                8, 
                24
            ], 
            "5": [
                4, 
                30, 
                33
            ], 
            "6": [
                20, 
                16, 
                21
            ], 
            "7": [
                21, 
                19, 
                7
            ], 
            "8": [
                11, 
[... 286 lines cut ...]
            "33": {
                "x": 21.281154632568359, 
                "y": 7.4636454582214355, 
                "z": 1.6519025564193726
            }, 
            "34": {
                "x": 20.048507690429688, 
                "y": 5.355278491973877, 
                "z": 2.3190517425537109
            }, 
            "35": {
                "x": 16.516220092773438, 
                "y": 6.3384995460510254, 
                "z": 2.4181685447692871
            }
        }
    }, 
    "datatype": "compas.datastructures/Mesh"
}

from compas.datastructures import Mesh
built_mesh = Mesh.from_data(mesh_json)
print(built_mesh.is_valid())
print(built_mesh.edges_on_boundaries())
```

brgcode, 2020-11-25:
will have look...

but just fyi.

`is_valid` is a misleading test, especially in this case. we will have top work on that. better to check `is_manifold` because that is where algorithms can fail...

closed 2020-12-01 by brgcode