compas-dev/compas #678: Suggest downgrading the Raise in DataDecoder's dtype to a printed warning...

tkmmark, 2020-11-25:
Because 'dtype' is such a common attribute name... a **raise** is somewhat excessive, especially in cases where a COMPAS-based developer may be using the DataDecoder to decode serialisable data from non-COMPAS sources...  Consider downgrading the raise for the **ValueError Exception** block at minimum, and introducing a **Else block** to let cls.from_data(o['value']) work its magic in the case that dtype does match the correct specification.

[image]

brgcode, 2020-12-01:
this is not an issue nor a feature request, but a PR in the making. please suggest code updates through PRs...

tkmmark, 2020-12-07:
ok

closed 2020-12-07 by tkmmark