compas-dev/compas #679: Suggestion (with code): Support for Tuples from DataEncoder and DataDecoder.

tkmmark, 2020-11-25:
The current DataEncoder and DataDecoders inherits the standard behaviour of the json.JSONEncoders/Decoders, which do not preserve Tuples (for dictionary values) during serialisation... causing Tuples to be cast as Lists when the serialised data is decoded.   Tuples, however, have widespread utility in COMPAS and in Python in general, and it is often desirable to maintain their datastructures... Since we are already in the habit of making the DataEncoder/Decoder great again, it would be nice to incorporate this feature into COMPAS' DataEncoders/Decoders as well:

This involves overriding the inherited **encode** method in the DataEncoder to introduce a tuples_encoder():
[image]

And then inserting the following to the **object_hook** in the DataDecoder:
[image]

For the convenience of COMPAS's developers, the insertions have been pasted below:
```
    def encode(self, o):

        def tuples_encoder(item):
            if isinstance(item, tuple):
                return {'__tuple__': True, 'items': item}
            elif isinstance(item, list):
                return [tuples_encoder(e) for e in item]
            elif isinstance(item, dict):
                return {key: tuples_encoder(value) for key, value in item.items()}
            else:
                return item

        return super(DataEncoder, self).encode(tuples_encoder(o))
```
```
        if '__tuple__' in o:

            return tuple(o['items'])
```

The author can confirm that this is compatible with the serialisation of core compas datastructures...
*The only issue that the author has observed with this modification is in compas_cloud.server's **load_cached**, where the following change will be needed:
[image]

brgcode, 2020-11-25:
for the convenience of the developers, a PR would be much better...

tkmmark, 2020-11-25:
Yes yes, the author will rebase his Compas soon to further convenience the developers.

closed 2020-12-07 by tkmmark