compas-dev/compas #685: FeatureRequest: use __str__ of type instead of type in run/import-time registration of object

tkmmark, 2020-12-01:
for artists, and rhino objects...
In order to make it more possible to use the current version of **unload_modules()** when working with a collection of modules and scripts (such as when one is developing plugin). since the equivalency of class/types will be gone for the same classes (even if no changes were made to the underlying classes) after invoking unload_modules().

gonzalocasas, 2020-12-01:
not sure where exactly you mean.. could you please add an example of this? is it related to the scene stuff?

tkmmark, 2020-12-07:
Ah yes, I meant situations like this:
[image]

tkmmark, 2020-12-07:
I was suggesting that we use some string based representation for building the type-to-object dict (e.g. __str__ of type, or dtype).  I realise this already sounds rather hacky, but what gave me the idea was when I was hoping to use unload_module to access updated code work done on a datastructure module, so I do not have to restart Rhino and reinitialise everything.  But it is rather tricky to get rid of all the old modules (between GH and IronPython, between IronPython and RunPythonScript) and the type for the same datastructure will not be equivalent as a result.

gonzalocasas, 2020-12-09:
I guess the best would be to state your root problem in the feature request, instead of a tentative implementation, otherwise, we might end up implementing a feature that does not solve your root problem in an optimal way.

Regarding what I understand so far as the root problem, it seems to be a matter of doing the right cleanup and potentially re-registration, rather than changing the internal representation of the registry.
If I get it right, you are developing a plugin,  and you want to improve your development workflow because certain things force you to reset the script engine to properly reload after changes. On a superficial level, I would say there's never the need to reset script engine for dev workflow, `unload_modules` (used with the right prefix to make sure everything we want is really unloaded) and probably a call to `register` afterwards sound like they could restore state to a clean situation.

However, I might be missing details about your workflow that make this more complicated. Could you please elaborate further on the workflow or attach a concrete example to replicate? (or a video?)