compas-dev/compas #701: RPC proxy not working with implicit start from Rhino/Grasshopper

tetov, 2020-12-11:
**Describe the bug**

Using `compas_rpc.Proxy`  as a contextmanager raises an exception. This only happens if a RPC server is not already running. (Same/similar to #265)

**To Reproduce**
Steps to reproduce the behavior:

**Context:** Grasshopper GHPython node or Rhino python editor.

## Sample script
```python
from compas.rpc import Proxy
from compas.geometry import Point

pt1 = Point(1, 2, 3)

with Proxy("compas.geometry", max_conn_attempts=5) as proxy:
    pt2 = proxy.Point(1, 2, 3)

if list(pt1) == pt2:
    print("Success")
```
4. Traceback
```
Message: Traceback (most recent call last):
  File "C:\Users\a\scoop\apps\miniconda3\4.7.12.1\envs\compas_rr_ts-dev\lib\site-packages\compas\rpc\dispatcher.py", line 100, in _dispatch
    module = importlib.import_module(modulename)
  File "C:\Users\a\scoop\apps\miniconda3\4.7.12.1\envs\compas_rr_ts-dev\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "C:\Users\a\scoop\apps\miniconda3\4.7.12.1\envs\compas_rr_ts-dev\lib\site-packages\compas\geometry\__init__.py", line 497, in <module>
    from .bbox import *  # noqa: F401 F403
  File "C:\Users\a\scoop\apps\miniconda3\4.7.12.1\envs\compas_rr_ts-dev\lib\site-packages\compas\geometry\bbox\__init__.py", line 10, in <module>
    from .bbox_numpy import *  # noqa: F401 F403
  File "C:\Users\a\scoop\apps\miniconda3\4.7.12.1\envs\compas_rr_ts-dev\lib\site-packages\compas\geometry\bbox\bbox_numpy.py", line 24, in <module>
    from compas.numerical import pca_numpy
  File "C:\Users\a\scoop\apps\miniconda3\4.7.12.1\envs\compas_rr_ts-dev\lib\site-packages\compas\numerical\__init__.py", line 109, in <module>
    from .isolines import *  # noqa: F401 F403
  File "C:\Users\a\scoop\apps\miniconda3\4.7.12.1\envs\compas_rr_ts-dev\lib\site-packages\compas\numerical\isolines\__init__.py", line 8, in <module>
    from .isolines_numpy import *  # noqa: F401 F403
  File "C:\Users\a\scoop\apps\miniconda3\4.7.12.1\envs\compas_rr_ts-dev\lib\site-packages\compas\numerical\isolines\isolines_numpy.py", line 11, in <module>
    import matplotlib.pyplot as plt
  File "C:\Users\a\scoop\apps\miniconda3\4.7.12.1\envs\compas_rr_ts-dev\lib\site-packages\matplotlib\__init__.py", line 174, in <module>
    _check_versions()
  File "C:\Users\a\scoop\apps\miniconda3\4.7.12.1\envs\compas_rr_ts-dev\lib\site-packages\matplotlib\__init__.py", line 159, in _check_versions
    from . import ft2font
ImportError: DLL load failed while importing ft2font: %1 is not a valid Win32 application.

Traceback:
  line 345, in _proxy, "C:\Users\a\AppData\Roaming\McNeel\Rhinoceros\6.0\scripts\compas\rpc\proxy.py"
  line 4, in <module>, "C:\Users\a\AppData\Local\Temp\TempScript.py"
```

**Expected behavior**
Print "Success".

**Desktop (please complete the following information):**
 - OS: Win10
 - Python version 2.7.8 (IronPython 2.7.8 (2.7.8.0) on .NET 4.0.30319.42000 (64-bit))
 - Python package manager: Anaconda
 - compas version: 0.19.1

brgcode, 2020-12-11:
does it work without the context mananger?

tetov, 2020-12-11:
```
from compas.rpc import Proxy
from compas.geometry import Point

pt1 = Point(1, 2, 3)

proxy = Proxy("compas.geometry", max_conn_attempts=5)
pt2 = proxy.Point(1, 2, 3)

if list(pt1) == pt2:
    print("Success")
```

Gives same traceback.

gonzalocasas, 2020-12-11:
ill take a look

gonzalocasas, 2020-12-11:
I just tested and it WOMM 😝
And btw, this stuff is part of the unit tests suite, so, in the general case, this should be working fine.

However, there's something about your environment that makes it fail, I'm just not sure what it might be. I see a few things that might be hints to figure out what happens:
- You are using a dev environment (based on the name), so the first thing I would try is to create a brand new env with compas 0.19.1, to rule out interference from other stuff
- You have miniconda instead of Anaconda, I don't think that is the source of the problem, but perhaps miniconda does take a different priority for some of the packages that normally are found on the anaconda channel (eg. matplotlib)
- Since the error seems to be coming from matplotlib, could you verify that 1) it works without RPC, 2) a `--force-reinstall` conda installation of the matplotlib-base package does not fix it ?
...

if none of that works, we'll need to figure out more things to test

tetov, 2020-12-11:
>     * You are using a dev environment (based on the name), so the first thing I would try is to create a brand new env with compas 0.19.1, to rule out interference from other stuff

That environment only had a compas 0.19.1 and an editable install of a package I tested compas.plugin with.

> * You have miniconda instead of Anaconda, I don't think that is the source of the problem, but perhaps miniconda does take a different priority for some of the packages that normally are found on [...]

Just tested this on another computer with miniconda and it worked. Seems like my plugin package is the culprit.

I'll investigate further, I'll reopen if necessary.

closed 2020-12-11 by tetov

tetov, 2020-12-14:
@gonzalocasas I'm still getting this. Installed Anaconda3 proper.
```
conda create -n repro6 compas==0.19.1 -y
conda activate repro6
python -m compas_rhino.uninstall
python -m compas_rhino.install
```
Then running above sample script gives the same error.

If I start rpc server manually (`compas_rpc start`) it works..

> Since the error seems to be coming from matplotlib, could you verify that 1) it works without RPC, 2) a --force-reinstall conda installation of the matplotlib-base package does not fix it ?
Tried but didn't fix.

brgcode, 2020-12-14:
is this on mac or windows?

tetov, 2020-12-14:
Windows!

brgcode, 2020-12-14:
ha sorry just saw the start of the thread. i guess this is related to the numpy issue on windows. we thought this was fixed but i forgot to update the conda recipe...

brgcode, 2020-12-14:
we have the same problem in RV2...

brgcode, 2020-12-14:
new windows builds are available for `0.19.1`. @tetov let me know if this solves the problem...

tetov, 2020-12-16:
@brgcode Hi! Tried it now, the latest version on conda-forge `win-64/compas-0.19.1-py38haa244fe_1.tar.bz2`. No change however.

brgcode, 2020-12-16:
could you check the version of numpy that was installed?

tetov, 2020-12-16:
`numpy 1.19.2 py38hdf1ac2f_1 conda-forge`

gonzalocasas, 2020-12-16:
Could it be you have a sandboxed install of conda via Scoop?

brgcode, 2020-12-16:
as far as i can tell it is related to this

https://developercommunity.visualstudio.com/content/problem/1207405/fmod-after-an-update-to-windows-2004-is-causing-a.html

however for us this went away by installing older numpy. maybe the same with mpl?

gonzalocasas, 2020-12-16:
could you please post your whole env variables (type `set` in your prompt with your environment activated)?

tetov, 2020-12-16:
Not scoop-ed anymore, this is standard anaconda install.
```
$ conda info
     active environment : repro7
    active env location : C:\Users\a\anaconda3\envs\repro7
            shell level : 5
       user config file : C:\Users\a\.condarc
 populated config files : C:\Users\a\.condarc
          conda version : 4.9.2
    conda-build version : 3.20.5
         python version : 3.8.5.final.0
       virtual packages : __cuda=11.0=0
                          __win=0=0
                          __archspec=1=x86_64
       base environment : C:\Users\a\anaconda3  (writable)
           channel URLs : https://conda.anaconda.org/conda-forge/win-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://repo.anaconda.com/pkgs/main/win-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/win-64
                          https://repo.anaconda.com/pkgs/r/noarch
                          https://repo.anaconda.com/pkgs/msys2/win-64
                          https://repo.anaconda.com/pkgs/msys2/noarch
          package cache : C:\Users\a\anaconda3\pkgs
                          C:\Users\a\.conda\pkgs
                          C:\Users\a\AppData\Local\conda\conda\pkgs
       envs directories : C:\Users\a\anaconda3\envs
                          C:\Users\a\.conda\envs
                          C:\Users\a\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/4.9.2 requests/2.24.0 CPython/3.8.5 Windows/10 Windows/10.0.19041
          administrator : False
             netrc file : None
           offline mode : False
```

tetov, 2020-12-16:
```
$ set
ALLUSERSPROFILE=C:\ProgramData
APPDATA=C:\Users\a\AppData\Roaming
ChocolateyInstall=C:\ProgramData\chocolatey
ChocolateyLastPathUpdate=132409340371986129
CommonProgramFiles=C:\Program Files\Common Files
CommonProgramFiles(x86)=C:\Program Files (x86)\Common Files
CommonProgramW6432=C:\Program Files\Common Files
COMPUTERNAME=UNIVERSAL-FRIEN
ComSpec=C:\Windows\system32\cmd.exe
CONDA_DEFAULT_ENV=repro7
CONDA_EXE=C:\Users\a\anaconda3\condabin\..\Scripts\conda.exe
CONDA_PREFIX=C:\Users\a\anaconda3\envs\repro7
CONDA_PREFIX_1=C:\Users\a\anaconda3
CONDA_PREFIX_2=C:\Users\a\anaconda3\envs\rcf
CONDA_PREFIX_3=C:\Users\a\anaconda3\envs\repro5
CONDA_PREFIX_4=C:\Users\a\anaconda3\envs\repro6
CONDA_PROMPT_MODIFIER=(repro7)
CONDA_PYTHON_EXE=C:\Users\a\anaconda3\python.exe
CONDA_SHLVL=5
DriverData=C:\Windows\System32\Drivers\DriverData
FOR_DISABLE_CONSOLE_CTRL_HANDLER=1
GEM_HOME=C:\Users\a\scoop\apps\ruby\current\gems
GEM_PATH=C:\Users\a\scoop\apps\ruby\current\gems
GIT_INSTALL_ROOT=C:\Users\a\scoop\apps\git\current
HOMEDRIVE=C:
HOMEPATH=\Users\a
JAVA_HOME=C:\Users\a\scoop\apps\adopt8-hotspot-jre\current
LOCALAPPDATA=C:\Users\a\AppData\Local
LOGONSERVER=\\UNIVERSAL-FRIEN
NUMBER_OF_PROCESSORS=4
OneDrive=C:\Users\a\OneDrive
OneDriveConsumer=C:\Users\a\OneDrive
OS=Windows_NT
Path=C:\Users\a\anaconda3\envs\repro7;C:\Users\a\anaconda3\envs\repro7\Library\mingw-w64\bin;C:\Users\a\anaconda3\envs\repro7\Library\usr\bin;C:\Users\a\anaconda3\envs\repro7\Library\bin;C:\Users\a\anaconda3\envs\repro7\Scripts;C:\Users\a\anaconda3\envs\repro7\bin;C:\Users\a\anaconda3\condabin;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0;C:\ProgramData\chocolatey\bin;C:\Windows\System32\OpenSSH;C:\Program Files\dotnet;C:\Program Files\Docker\Docker\resources\bin;C:\ProgramData\DockerDesktop\version-bin;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit;C:\Users\a\scoop\apps\gpg4win\3.1.13\GnuPG\bin;C:\Users\a\scoop\apps\adopt8-hotspot-jre\current\bin;C:\Users\a\scoop\apps\ruby\current\bin;C:\Users\a\scoop\apps\ruby\current\gems\bin;C:\Users\a\scoop\apps\gpg4win\current\Gpg4win\bin;C:\Users\a\scoop\apps\gpg4win\current\GnuPG\bin;C:\Users\a\scoop\shims
PATHEXT=.COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC
PROCESSOR_ARCHITECTURE=AMD64
PROCESSOR_IDENTIFIER=Intel64 Family 6 Model 142 Stepping 9, GenuineIntel
PROCESSOR_LEVEL=6
PROCESSOR_REVISION=8e09
ProgramData=C:\ProgramData
ProgramFiles=C:\Program Files
ProgramFiles(x86)=C:\Program Files (x86)
ProgramW6432=C:\Program Files
PROMPT=(repro7) $P$G
PSModulePath=C:\Program Files\WindowsPowerShell\Modules;C:\Windows\system32\WindowsPowerShell\v1.0\Modules
PUBLIC=C:\Users\Public
SESSIONNAME=Console
SystemDrive=C:
SystemRoot=C:\Windows
TEMP=C:\Users\a\AppData\Local\Temp
TMP=C:\Users\a\AppData\Local\Temp
USERDOMAIN=UNIVERSAL-FRIEN
USERDOMAIN_ROAMINGPROFILE=UNIVERSAL-FRIEN
USERNAME=a
USERPROFILE=C:\Users\a
VS140COMNTOOLS=C:\Program Files (x86)\Microsoft Visual Studio 14.0\Common7\Tools\
windir=C:\Windows
```

brgcode, 2020-12-16:
could it be because of the powershell? is it possible that when trying to launch the server from Rhino/GH that this goes through powershell instead of the normal prompt? VS Code also needs to be configured to not do that...

tetov, 2020-12-16:
I'm using cmd.exe for conda through the provided shortcuts, Powershell has not been involved in any of this..

brgcode, 2020-12-16:
no i understand but could it be that that is what is being launched by the Rhino script? is there a way to check which shell is being used by the subprocess?

tetov, 2020-12-16:
```
from subprocess import Popen

a = Popen(["if", "%COMSPEC%", "NEQ", "C:\Windows\system32\cmd.exe", "exit", "1"], shell=True)

a.wait()

print(a.returncode)
```

Quick test. I get return code zero.

gonzalocasas, 2020-12-16:
The issue is probably connected to how we rebuild the environment (`compas._os.prepare_environment()`). There is something in your env that is different from what we do, causing the proxy to have incomplete env setup and fail.

gonzalocasas, 2020-12-16:
could you please post the latest stack track with the new environment you created (the one with anaconda and repro7)?

tetov, 2020-12-16:
> could you please post the latest stack track with the new environment you created (the one with anaconda and repro7)?

```
Message: Traceback (most recent call last):
  File "C:\Users\a\anaconda3\envs\repro7\lib\site-packages\compas\rpc\dispatcher.py", line 100, in _dispatch
    module = importlib.import_module(modulename)
  File "C:\Users\a\anaconda3\envs\repro7\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "C:\Users\a\anaconda3\envs\repro7\lib\site-packages\compas\geometry\__init__.py", line 497, in <module>
    from .bbox import *  # noqa: F401 F403
  File "C:\Users\a\anaconda3\envs\repro7\lib\site-packages\compas\geometry\bbox\__init__.py", line 10, in <module>
    from .bbox_numpy import *  # noqa: F401 F403
  File "C:\Users\a\anaconda3\envs\repro7\lib\site-packages\compas\geometry\bbox\bbox_numpy.py", line 24, in <module>
    from compas.numerical import pca_numpy
  File "C:\Users\a\anaconda3\envs\repro7\lib\site-packages\compas\numerical\__init__.py", line 109, in <module>
    from .isolines import *  # noqa: F401 F403
  File "C:\Users\a\anaconda3\envs\repro7\lib\site-packages\compas\numerical\isolines\__init__.py", line 8, in <module>
    from .isolines_numpy import *  # noqa: F401 F403
  File "C:\Users\a\anaconda3\envs\repro7\lib\site-packages\compas\numerical\isolines\isolines_numpy.py", line 11, in <module>
    import matplotlib.pyplot as plt
  File "C:\Users\a\anaconda3\envs\repro7\lib\site-packages\matplotlib\__init__.py", line 174, in <module>
    _check_versions()
  File "C:\Users\a\anaconda3\envs\repro7\lib\site-packages\matplotlib\__init__.py", line 159, in _check_versions
    from . import ft2font
ImportError: DLL load failed while importing ft2font: %1 is not a valid Win32 application.

Traceback:
  line 345, in _proxy, "C:\Users\a\AppData\Roaming\McNeel\Rhinoceros\6.0\scripts\compas\rpc\proxy.py"
  line 7, in <module>, "C:\Users\a\AppData\Local\Temp\TempScript.py"
```

gonzalocasas, 2020-12-16:
thx! I find the content of `CONDA_EXE` in your setup very strange (`CONDA_EXE=C:\Users\a\anaconda3\condabin\..\Scripts\conda.exe`),
could you try editing `compas._os.prepare_environment()` and change the following:

```python
    if CONDA_EXE:
        env['CONDA_EXE'] = CONDA_EXE
```

for
```python
    if CONDA_EXE:
        env['CONDA_EXE'] = r'C:\Users\a\anaconda3\python.exe'
```
?

tetov, 2020-12-16:
Nothing changed I'm afraid.

This conda install was as standard as it gets, just latest release without any changes.. The `%USERPROFILE%/anaconda3` folder didn't exist before installing so it's not a leftover either..

brgcode, 2020-12-16:
how come conda is on the path? is this done on the fly by the anaconda prompt or did you choose to add it during installation?

tetov, 2020-12-16:
Reinstalled Anaconda since it seems like I might have ticked the box "Add anaconda to path". Still have `C:\Users\a\anaconda3\condabin;` on there. Next I'll try to edit the path manually.

```Path=
C:\Users\a\anaconda3\envs\repro8;
C:\Users\a\anaconda3\envs\repro8\Library\mingw-w64\bin;
C:\Users\a\anaconda3\envs\repro8\Library\usr\bin;
C:\Users\a\anaconda3\envs\repro8\Library\bin;
C:\Users\a\anaconda3\envs\repro8\Scripts;
C:\Users\a\anaconda3\envs\repro8\bin;
C:\Users\a\anaconda3\condabin;
C:\Windows\system32;C:\Windows;
C:\Windows\System32\Wbem;
C:\Windows\System32\WindowsPowerShell\v1.0;
C:\ProgramData\chocolatey\bin;
C:\Windows\System32\OpenSSH;
C:\Program Files\dotnet;
C:\Program Files\Docker\Docker\resources\bin;
C:\ProgramData\DockerDesktop\version-bin;
C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit;
C:\Users\a\scoop\apps\gpg4win\3.1.13\GnuPG\bin;
C:\Users\a\scoop\apps\adopt8-hotspot-jre\current\bin;
C:\Users\a\scoop\apps\ruby\current\bin;
C:\Users\a\scoop\apps\ruby\current\gems\bin;
C:\Users\a\scoop\apps\gpg4win\current\Gpg4win\bin;
C:\Users\a\scoop\apps\gpg4win\current\GnuPG\bin;
C:\Users\a\scoop\shims
```

tetov, 2020-12-16:
Seems like all conda directories on PATH is set per terminal session.

brgcode, 2020-12-16:
if you reinstall Anaconda a full PC restart might be appropriate...

tetov, 2020-12-16:
> if you reinstall Anaconda a full PC restart might be appropriate...

Ok! I'll see you on the other side ;).
[image]

tetov, 2020-12-16:
> thx! I find the content of `CONDA_EXE` in your setup very strange (`CONDA_EXE=C:\Users\a\anaconda3\condabin\..\Scripts\conda.exe`),

After fresh reinstall and reboot there's still a discrepancy here (just thought I'd mention it): 
```
$ set
[...]
CONDA_EXE=C:\Users\a\anaconda3\Scripts\conda.exe
[...]
CONDA_PYTHON_EXE=C:\Users\a\anaconda3\python.exe
[...]
```

brgcode, 2020-12-16:
but does it work now?

tetov, 2020-12-16:
Oh, sorry. Nope.

tetov, 2020-12-16:
Have you been able to reproduce at all?

Win 10, fresh conda environment, sample code from start of thread?

gonzalocasas, 2020-12-16:
No repro so far. As mentioned, this case is part of the unit tests so a complete clean setup does not exhibit this issue. I also tried my machine, and still no repro. It's something related to your setup, but you're probably not the only one, so it would be interesting to try to locate the problem. How about a zoom call?

tetov, 2020-12-16:
Sure, send me an invite when it works for you. :)

brgcode, 2020-12-16:
i could reproduce the problem just now. not sure why i couldn't before...

i think the problem is related to the order in which things are added to `PATH` by the function `compas._os.prepare_environment()`

when you run this from the terminal the paths related to the current environment are already on `PATH` and quite to the front. when you run this from Rhino `PATH` obviously doesn't contain those environment paths yet and then they are added to the back.

it is the only difference i can see between the two situations.

i created an environment with a from source install and modified the following in `compas._os.prepare_environment()` such also in Rhino the paths of the conda environment are in the front

```python
    if PYTHON_DIRECTORY:
        lib_bin = os.path.join(PYTHON_DIRECTORY, 'Library', 'bin')
        if os.path.exists(lib_bin) and lib_bin not in env['PATH']:
            env['PATH'] = lib_bin + os.pathsep + env['PATH']

        lib_bin = os.path.join(PYTHON_DIRECTORY, 'lib')
        if os.path.exists(lib_bin) and lib_bin not in env['PATH']:
            env['PATH'] = lib_bin + os.pathsep + env['PATH']
```

this solves the problem for me. however it could also be that it is just because of the install from source :)
will test a few more install scenarios to rule that out and send a PR...

brgcode, 2020-12-16:
also, if the two cases in the snippet above are meant to cover two types of OS (the first Windows and the second Unix) the contents of those two folders are not the same...

gonzalocasas, 2020-12-16:
@brgcode ah! that makes a lot of sense. if the `PATH` has stuff that would screw up finding the correct lib/bins, prepending instead of appending would solve.

brgcode, 2020-12-16:
the commad line also adds quite a bit more items to the beginning of `PATH`

```
PATH C:\Users\vanmelet\Anaconda3\envs\testrpc;C:\Users\vanmelet\Anaconda3\envs\testrpc\Library\mingw-w64\bin;C:\Users\vanmelet\Anaconda3\envs\testrpc\Library\usr\bin;C:\Users\vanmelet\Anaconda3\envs\testrpc\Library\bin;C:\Users\vanmelet\Anaconda3\envs\testrpc\Scripts;C:\Users\vanmelet\Anaconda3\envs\testrpc\bin;
```

the `prepare_environment` function only adds

```
PATH C:\Users\vanmelet\Anaconda3\envs\testrpc2\lib;C:\Users\vanmelet\Anaconda3\envs\testrpc2\Library\bin;
```

brgcode, 2020-12-16:
on Mac in the terminal you get 
```
PATH /Users/vanmelet/anaconda3/envs/compas-dev/bin:/Users/vanmelet/anaconda3/condabin:
```
in Rhino
```
PATH /Users/vanmelet/anaconda3/envs/compas-dev/lib
```
the latter is the result of `compas._os.prepare_environment` but is not equivalent to the folder add by the function on windows.
equivalent would be 

1. `C:\Users\vanmelet\Anaconda3\envs\testrpc2\Library\bin` and `/Users/vanmelet/anaconda3/envs/compas-dev/bin`
2. `C:\Users\vanmelet\Anaconda3\envs\testrpc2\Library\lib` and `/Users/vanmelet/anaconda3/envs/compas-dev/lib`

tetov, 2020-12-16:
> ```python
>     if PYTHON_DIRECTORY:
>         lib_bin = os.path.join(PYTHON_DIRECTORY, 'Library', 'bin')
>         if os.path.exists(lib_bin) and lib_bin not in env['PATH']:
>             env['PATH'] = [...]
This fixed it for me.