compas-dev/compas #702: More ways to advertise plugins

tetov, 2020-12-11:
# Feature Request

As a extension/plugin dev, I want a more flexible way to advertise plugin functionality.

## Details

Currently plugins are searched for in packages named `compas_*`. My current project's fab package is named gramaziokohler/rapid_clay_formations_fab. For a bit it was named `compas_rcf` but this was changed since:

1. It's not meant for general users in the compas ecosystem. 
2. No one knows what RCF stands for. 

**Describe the solution you'd like**
Being able to advertise plugins in `setup.py`s `entrypoints` or something similar. 

https://setuptools.readthedocs.io/en/latest/userguide/entry_point.html

tetov, 2020-12-11:
(I want to advertise a rhino_installable_package`)

gonzalocasas, 2021-02-01:
My first implementation indeed relied on `setuptools` and entrypoint hooks, however, I had to revert to a naming-convention style of discovery because plugins should also be discoverable inside `IronPython` which totally lacks `setuptools` infrastructure.

I realize it's a bit inconvenient, however, notice there's a simple workaround: you don't have to rename your entire project to be `compas_xyz`, you can keep your name but include a top-level package called `compas_rcf_plugins` (or whatever fits your case), which contains the entry points to plugins for the compas framework.

tetov, 2021-03-24:
I implemented this for a [package I wrote yesterday](https://github.com/tetov/mmec_fab) and it works well, in case anyone wants to see an example.

```
mmec_fab
├── README.md
├── setup.cfg
├── setup.py
└── src
    ├── compas_mmec_fab_plugin
    │   └── __init__.py
    └── mmec_fab
        ├── __init__.py
        ├── module1.py
        ├── module2.py
        └── rhino_install.py
```

`src/compas_mmec_fab_plugin/__init__.py`

```python
__all_plugins__ = ["mmec_fab.rhino_install"]
```

`src/mmec_fab/rhino_install.py`
```python
import compas.plugins

@compas.plugins.plugin(category="install")
def installable_rhino_packages():
    return ["mmec_fab", "compas_rrc"]
```
(I included compas_rrc here as well since it hasn't got a `installable_rhino_packages()` plugin.)

jf---, 2024-04-08:
@gonzalocasas your comment seems a reasonable closing statement?

closed 2024-04-09 by tetov