compas-dev/compas #703: Online documentation contains html dump

Achillx, 2020-12-11:
**Describe the bug**
The online API documentation page displays HTML code.

**To Reproduce**
Visit [https://compas.dev/compas/latest/api/compas.datastructures.html](https://compas.dev/compas/latest/api/compas.datastructures.html)

**Expected behavior**
Well...no HTML dumps :p

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: macOS Catalina 10.15.7
 - Chrome: Version 87.0.4280.88 (Official Build) (x86_64)
 - Safari: Version 14.0.1 (15610.2.11.51.10, 15610)

beverlylytle, 2020-12-11:
Same with geometry https://compas.dev/compas/latest/api/compas.geometry.html

brgcode, 2020-12-11:
this is the inheritance diagram that is somehow not converted to an image by `graphviz`. will have a look...

tetov, 2020-12-14:
I was looking at this a bit yesterday.

Two things:
* You need to install both python-graphviz and graphviz on the github runnner. `sudo apt-get install graphviz`. 
* Uncomment graphviz extension in `docs/conf.py`. 

Tested locally but not on github action-like environment.

Licini, 2020-12-14:
> I was looking at this a bit yesterday.
> 
> Two things:
> 
> * You need to install both python-graphviz and graphviz on the github runnner. `sudo apt-get install graphviz`.
> * Uncomment graphviz extension in `docs/conf.py`.
> 
> Tested locally but not on github action-like environment.

Thanks! gonna give a try now on github action

closed 2020-12-14 by brgcode