compas-dev/compas #704: BaseMesh attributes documentation is wrong

Achillx, 2020-12-11:
**Describe the bug**
The [online documentation](https://compas.dev/compas/latest/api/generated/compas.datastructures.BaseMesh.html#compas.datastructures.BaseMesh), shows `max_int_key` attributes that do not exist in `Mesh.data` dictionary. Also accessing the `data` attribute returns a dict that contains `compas` `datatype` and `data` (again!)

**To Reproduce**
Run the following code:
```
from compas.geometry import Box
from compas.datastructures import Mesh

box = Box.from_width_height_depth(2, 2, 2)
box = Mesh.from_shape(box)
print(box.name)
print(box.data.keys())
print(box.data['data'].keys())
print(box.data['data']['max_face'])
# The line bellow will fail
print(box.data['max_face'])
```

**Expected behavior**
Get for example `max_face` but calling `Mesh.data['max_face']`

**Desktop (please complete the following information):**
 - OS: macOS Catalina 10.15.7
 - Python version: 3.7.6
 - Python package manager : conda
 - Compas: 0.91.1

brgcode, 2020-12-11:
left-over from old code. will remove it...

brgcode, 2020-12-14:
solved since 3cf2bd98

closed 2020-12-14 by brgcode

Achillx, 2020-12-14:
I would also add this to the css styles:
```
.graphviz img {
    width: 100%;
}
```

It would fix the overflow issue seen below.
[image]

brgcode, 2020-12-14:
already did, refresh page...

Achillx, 2020-12-14:
One more comment.
Personally and maybe for others, is very useful to get the last face or vertex index. Sure we can get it through `Mesh.data['data']['max_vertex']`, but now that the whole `data` dictionary is not documented it is very much hidden and a bit counter-intuitive that you need to access data twice. I'm not saying it is a bug, but it would be nice to have another attribute for `data` and be able to call `Mesh.data['max_vertex']`.

brgcode, 2020-12-14:
the data dict is not meant to be used this way. perhaps we should underscore the attribute name. `max_vertex` is an attribute of the mesh and can be consulted at all times. will assign a property to this...