compas-dev/compas #705: Conversion functions between CAD geometry objects and compas geometry objects, part 2

tetov, 2020-12-11:
# Feature Request

I brought this up some time ago in #458.

I've kept converting between `compas.geometry` and `Rhino.Geometry` in my grasshopper files, mostly to visualize results.

The solution brought up in #458 takes the form of statements like this:
```
cgmesh = RhinoMesh.from_geometry(rgmesh).to_compas()
```
or a bit more verbose:
```
wrapper = RhinoMesh.from_geometry(rgmesh)
cgmesh = wrapper.to_compas()
```

Now this makes a lot of sense to users who have a good understanding of object oriented programming.

However, I've found it to be harder for beginners to grasp. (What's a wrapper object? What's a class method? etc..)

I've written code since I was 12 as a hobby and it was only a few years ago I really understood the OO paradigm.

For the benefit of new users, who might be new to both compas and programming I think we should offer conversion functions alongside the wrappers in `compas_rhino.geometry` and potentially `compas_blender.geometry`.

Here are some example of [compas to rhino conversions](https://github.com/gramaziokohler/rapid_clay_formations_fab/blob/master/src/rapid_clay_formations_fab/rhino/compas_to_rhino.py) and [vice versa](https://github.com/gramaziokohler/rapid_clay_formations_fab/blob/master/src/rapid_clay_formations_fab/rhino/rhino_to_compas.py).

tetov, 2021-06-18:
I packaged my functions together with a universal converter and a Grasshopper node in [compas_convert](https://github.com/biodigitalmatter/compas_convert). :)

brgcode, 2021-11-09:
i don't think there is an actual issue anymore here. most geometry types of Rhino are no wrapped with corresponding conversion and coercion functions...

closed 2021-11-09 by brgcode